telmomarques/xiaomi-360-1080p-hacks #3: Compile suitable RTSP server

telmomarques, 2019-02-21:
Choose a suitable (lightweight, secure) RTSP server and compile it for ARMv7, if needed.

vladimirovsan, 2019-04-01:
hello, please help configure the RTSP stream for vMix

telmomarques, 2019-04-01:
Hi @vladimirovsan, there's still no RTSP server. This issue is to track the future development of a server.

vladimirovsan, 2019-04-01:
"Username is "root" (no quotes), and there is no password"

root - 
[image]
I get incorrect login

telmomarques, 2019-04-01:
It should not ask for password... it should go directly from the login prompt to the shell.
Make sure you're not adding spaces or any other characters to the username. Do not use backspace, also. Write the username correctly in one go and hit enter.

ChoiZ, 2019-10-20:
Hi, any news about the RTSP server?

telmomarques, 2019-10-21:
Hi,

You can follow the telegram group on https://t.me/mijav4RTSP for more info

none815, 2020-01-03:
I've uploaded the source posted by P on Telegram [on this repository](https://github.com/none815/live555) to make it easier accessible for compiling/development.

SedrK, 2020-01-03:
Is that now working?