telmomarques/xiaomi-360-1080p-hacks #6: [New Hack] HTTP server

telmomarques, 2019-02-21:
Choose a suitable (lightweight, secure) HTTP server.
Initially to serve as a base for #5.

j0se, 2019-04-03:
I think that HTTP server needs, php and sqlite support what do you think ? ( For now I only have the web server :) )

telmomarques, 2019-04-03:
To be honest I still haven't though about that too much, but initially the ideia is to provide a GUI for configuration files. So, on the configuration context, the persistence should be done in some sort of json or yaml files (one for "hack"), so they can also be easy modified in any text editor.

Other advanced functionalities may need some server side language and database, but I still haven't though of that. Any lightweight solutions should be OK, though.

funivan, 2019-05-02:
You can use Lighttpd + sh script (or python, ruby etc)  to handle request data.

ppisljar, 2019-09-30:
i tried running nodejs (precompiled for armv7), seems to work fine. I will try to use express web server with it.

telmomarques, 2019-09-30:
Hey,

I have some reservations about using relatively heavy webservers, as node. The SoC reaches loads above 1 very easily, and the complexity we need to add might not justify a heavy framework.

telmomarques, 2019-09-30:
> i tried running nodejs (precompiled for armv7), seems to work fine. I will try to use express web server with it.
> 
> how would i go forward with installing this ?
> 
> * i created /bin and /lib folder on sdcard and put node files in there
> * i added a script under /x306h/httpd which:
> * adds /mnt/sdcard/bin to the PATH
> * adds /mnt/sdcard/lib to LD_LIBRARY_PATH
> 
> however now when i try `node myindex.js` i get seg fault ...
> but if i do `/mnt/sdcard/bin/node myindex.js` it works ok ...
> 
> any idea what that could be for ?

Try to strace it, you can find a static strace binary for arm here: https://github.com/andrew-d/static-binaries/blob/master/binaries/linux/arm/strace

ppisljar, 2019-09-30:
thanks!

i agree we shouldn't use this in the long run i was just playing with it ... anyway its super super slow.

busybox includes a lightweight httpd, which should be sufficient (its used in some hacks for other xiaomi cameras)