telmomarques/xiaomi-360-1080p-hacks #7: Support newer firmwares

telmomarques, 2019-02-22:
Newer firmwares disable the S49factory exploit. Another exploit needs to be found for newer firmware versions.

none815, 2020-01-02:
Might be possible to manually flash the newer firmware and modify S49factory in the process to keep the exploit working.
(Disclaimer: i haven't tested this, but maybe someone who can recover from a brick could give it a try.)

- Download current [firmware 3.4.5_008](http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc016/firmware/IPC016_16.3.4.5_0081.zip)
- Extract firmware files:
```
dd if=tf_recovery.img of=kernel.bin bs=1 count=2162688
dd if=tf_recovery.img of=rootfs.bin bs=1 count=7667712 skip=2162688
dd if=tf_recovery.img of=data.bin bs=1 count=6488064 skip=9830400
```
- Flash the new firmware:
```
dd if=/mnt/sdcard/kernel.bin of=/dev/mtd1
dd if=/mnt/sdcard/rootfs.bin of=/dev/mtd2
dd if=/mnt/sdcard/data.bin of=/dev/mtd3
```
- Modify the S49factory file:

>  /etc/init.d/S49factory

```
#!/bin/sh 
/mnt/sdcard/override.sh
```

elebeaup, 2020-01-10:
> It is possible to manually flash the newer firmware and modify the data partition to load a custom script.
> 
> * Download current [firmware 3.4.5_008](http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc016/firmware/IPC016_16.3.4.5_0081.zip)
> * Extract firmware files:
> 
> ```
> dd if=tf_recovery.img of=kernel.bin bs=1 count=2162688
> dd if=tf_recovery.img of=rootfs.bin bs=1 count=7667712 skip=2162688
> dd if=tf_recovery.img of=data.bin bs=1 count=6488064 skip=9830400
> ```
> 
> * Patch the data partition:
> 
> ```
> #!/bin/bash
> # patch jffs2 partition
> sudo modprobe mtdblock
> sudo modprobe mtdram total_size=6336
> sudo dd if=data.bin of=/dev/mtdblock0 bs=1
> mkdir mount
> sudo mount -t jffs2 /dev/mtdblock0 mount
> echo '#!/bin/sh' | sudo tee -a mount/bin/log_diag_platform.sh
> echo '/mnt/sdcard/override.sh' | sudo tee -a mount/bin/log_diag_platform.sh
> sudo chmod 755 mount/bin/log_diag_platform.sh
> sudo umount mount
> rmdir mount
> sudo dd if=/dev/mtdblock0 of=data.bin bs=1
> ```
> 
> * Flash the new firmware:
> 
> ```
> dd if=/mnt/sdcard/kernel.bin of=/dev/mtdblock1
> dd if=/mnt/sdcard/rootfs.bin of=/dev/mtdblock2
> dd if=/mnt/sdcard/data.bin of=/dev/mtdblock3
> ```
> 
> * SDcard script:
> 
> > /sdcard/override.sh
> 
> ```
> #!/bin/sh
> main() {
>   # start telnet
>   /mnt/sdcard/busybox telnetd
> }
> 
> if [ ! -f /tmp/.override ]; then
>  touch /tmp/.override
>  main
> fi
> ```

The data partition is both /dev/mtdblock3 and /dev/mtdblock0 ?

none815, 2020-01-10:
@elebeaup  
The first two steps are for preparation and doesn't happen on the camera, only the flashing process needs to be done on the camera.

rezmus, 2020-02-18:
@none815 "Copy the new firmware to the sdcard and flash it to the camera via telnet/ssh"

you flash firmware via telnet/ssh to enable telnet or i'm missing something here? ;)

elebeaup, 2020-02-19:
@rezmus I think you must downgrade the camera firmware to 3.4.2_0062 and install the hacks and after you can upgrade to a new firmware release via telnet

rezmus, 2020-02-19:
are you sure? this github hack was done for ipc009 camera (MJSXJ05CM), and firmware from this thread is for ipc016 (CMSXJ16A). imi/chuangmi has ~20 cams (~10 360/1080p) and soc in any of them may be same or different to ipc009.

telmomarques, 2020-02-19:
Steps are:
1. Downgrade firmware
2. Install the hacks
3. Use the hacks to access camera via telnet
4. Flash newer firmware (patched) using dd

You just need to make sure to have the correct newer firmware (patched) for your camera.

nhatnamg, 2020-03-16:
I have found 2 versions available for IPC009 model:
http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc009/firmware/IPC009_3.4.6_0213.zip
http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc009/firmware/IPC009_3.5.1_0217.zip

Currently when checking my camera with Mi Home app, it shows that the newest version is 3.4.6_213. So I decided to used this version. On the first attempt, the camera brick when I was flashing rootfs.bin (don't know why). I have to recover the camera with the vulnerable version 3.4.2_0062. On the second attempt, I sleeped the camera first. The flashing process went well. The camera operates normally with the patched firmware. No more update prompted on the app.
I didn't confirm the 3.5.1_0217 version yet but I've checked the file systems. It seem they have updated some parts but not much changes.

nhatnamg, 2020-03-19:
Update: I will NOT recommend using version 3.5.1_0217. It has some minor updates but some script files in /etc/init.d/ will try to "umount /mnt/sdcard" upon finishing. Therefore if you place your hack scripts on /mnt/sdcard, they won't work. So to use 3.5.1_0217, you have to remove all "umount /mnt/sdcard" points in its init.d scripts.

Apparently there is 2 ways for patching firmware:
1. The way @elebeaup did: inject your codes into <data.bin>/bin/log_diag_platform.sh. It will work but normally you will have to wait about 5 minutes after camera restarted because this script is executed by crontab on 5 mins interval.
2. Instead of patching data partition, you can patch rootfs to make it behaves just like version 3.4.2_0062

```
apt-get install squashfs-tools
unsquashfs rootfs.bin

# Replace squashfs-root/etc/init.d/S49factory with the same one on version 3.4.2_0062

mksquashfs squashfs-root rootfs_patched.bin -comp xz
```

**/etc/init.d/S49factory** on version **3.4.2_0062**
```
#!/bin/sh

#must to verify the sdcard

if [ -f /mnt/sdcard/manu_test/manu.sh ]
then
touch /tmp/factory_mode
/mnt/sdcard/manu_test/manu.sh
fi
```
then flash rootfs_patched.bin to /dev/mtdblock2 of the camera.

kathgironpe, 2020-05-24:
> Steps are:
> 
> 1. Downgrade firmware
> 2. Install the hacks
> 3. Use the hacks to access camera via telnet
> 4. Flash newer firmware (patched) using dd
> 
> You just need to make sure to have the correct newer firmware (patched) for your camera.
> 
> P.S. as of this moment I haven't provided a patched latest version of the firmware for the MJSXJ02CM, as I didn't work on this subject yet.

Trying my luck with this because the camera goes black without me upgrading via Mi Home app.

KhArtNJava, 2020-05-25:
> It is possible to manually flash the newer firmware and modify the data partition to load a custom script.
> 
> * Shell script to download and patch the current firmware:
> 
> ```
> #!/bin/bash
> # download firmware
> wget http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc016/firmware/IPC016_16.3.4.5_0081.zip
> unzip -j IPC016_16.3.4.5_0081.zip
> rm IPC016_16.3.4.5_0081.zip
> 
> # extract firmware content
> mkdir -p firmware
> dd if=tf_recovery.img of=firmware/kernel.bin bs=1 count=2162688
> dd if=tf_recovery.img of=firmware/rootfs.bin bs=1 count=7667712 skip=2162688
> dd if=tf_recovery.img of=firmware/data.bin bs=1 count=6488064 skip=9830400
> rm tf_recovery.img
> 
> # patch jffs2 partition
> sudo modprobe mtdblock
> sudo modprobe mtdram total_size=6336
> sudo dd if=firmware/data.bin of=/dev/mtdblock0 bs=1
> mkdir mount
> sudo mount -t jffs2 /dev/mtdblock0 mount
> echo '#!/bin/sh' | sudo tee -a mount/bin/log_diag_platform.sh
> echo '/mnt/sdcard/override.sh' | sudo tee -a mount/bin/log_diag_platform.sh
> sudo chmod 755 mount/bin/log_diag_platform.sh
> sudo umount mount
> rmdir mount
> sudo dd if=/dev/mtdblock0 of=firmware/data.bin bs=1
> ```
> 
> * Copy the new firmware to the sdcard and flash it to the camera via telnet/ssh:
> 
> ```
> dd if=/mnt/sdcard/kernel.bin of=/dev/mtdblock1
> dd if=/mnt/sdcard/rootfs.bin of=/dev/mtdblock2
> dd if=/mnt/sdcard/data.bin of=/dev/mtdblock3
> ```
> 
> * SDcard script:
> 
> > /sdcard/override.sh
> 
> ```
> #!/bin/sh
> if [ ! -f /tmp/.override ]; then
>     touch /tmp/.override
> 	
>     # start telnet server
>     /mnt/sdcard/busybox telnetd
> fi
> ```

none815, how did you find count values for dd?
> dd if=tf_recovery.img of=firmware/kernel.bin bs=1 count=2162688
> dd if=tf_recovery.img of=firmware/rootfs.bin bs=1 count=7667712 skip=2162688
> dd if=tf_recovery.img of=firmware/data.bin bs=1 count=6488064 skip=9830400

How to get these values for this https://drive.google.com/file/d/1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj/view firmware? 
What is algorithm (logic steps) to obtain these values for that img file?

Thank you in advance.

KhArtNJava, 2020-05-25:
Oh, looks like I see, that it's only 2+7+6 MB. Cool. Thank you for sharing this.
But MJSXJ05CM doesn't have root access, looks like SSH/telnet port is closed on it. How to access SSH/telnet on MJSXJ05CM?

KhArtNJava, 2020-05-25:
Oh!!!
I understand it now https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/18 . Cool, I'm going to AliExpress for a programmer and clips )))

KhArtNJava, 2020-05-25:
Anyway, @none815, how did you discovered sizes 2162688, 7667712 and 6488064? Any marks in *.img files? Or, may be some of volume sizes standard?

rezmus, 2020-05-25:
# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00050000 00010000 "BOOT"
mtd1: 00210000 00010000 "KERNEL"
mtd2: 00750000 00010000 "ROOTFS"
mtd3: 00630000 00010000 "DATA"
mtd4: 00010000 00010000 "CONFIG"
mtd5: 00010000 00010000 "FACTORY"

KhArtNJava, 2020-05-25:
> cat /proc/mtd
> dev: size erasesize name
> mtd0: 00050000 00010000 "BOOT"
> mtd1: 00210000 00010000 "KERNEL"
> mtd2: 00750000 00010000 "ROOTFS"
> mtd3: 00630000 00010000 "DATA"
> mtd4: 00010000 00010000 "CONFIG"
> mtd5: 00010000 00010000 "FACTORY"

Thank you.
But in your case there's no value 2162688 for kernel, it's 00210000. Are your results from different device or maybe I can't understand meaning of value 00210000 for kernel?

KhArtNJava, 2020-05-25:
And, okay, how can I get those values, if I can't run cat /proc/mtd because there's no ssh/telnet access on MJSXJ05CM? Yeah, I know values for MJSXJ05CM now, but let's imagine that there's another one camera, without ssh/telnet and I can only access flash chip of the camera. How can I get those values to split dumped with dd img file from flash chip?

mwkchan, 2020-05-26:
This is the latest firmware http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc009/firmware/IPC009_3.5.8.0406.zip

I tired to mount rootfs.bin to modify S49Factory but mount -t jffs2 /dev/mtdblock0 gave me a "Can't read superblock" error.

rezmus, 2020-05-26:
> But in your case there's no value 2162688 for kernel, it's 00210000. Are your results from different device or maybe I can't understand meaning of value 00210000 for kernel?

210000 hex = 2162688 dec. flash layout is the same for all imilab cams. just follow https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/18 to get telnet access.

mwkchan, 2020-05-28:
> rootfs is squashfs, data is jffs2.

I see. That means I can't mount it rw and modify it. I tried unsquash it and modify it. After I squashed it back and dded it back, the camera was bricked. Luckily a firmware downgrade using an SD Card unblocked it.

Cseb62, 2020-05-29:
Hi could You tell me what files to put on the sdcard ? thanks

klym-tatarnikov, 2020-06-07:
Hi.
I've tried to use 'manual' update of IPC016 with new version of FW but something went wrong and now I've bricked the device... It looks U-boot was overwritten. 
I have dump of all MTD blocks. But U-boot is not there.
If someone have fullflash dump file could please share it contents (fist 0x50000 bytes should be u-boot)

rezmus, 2020-06-07:
if you did backup of /dev/mtdblock0 you should have uboot. anyway here is my copy.

https://ufile.io/rsb6u7fa

Cseb62, 2020-06-08:
> si vous avez fait une sauvegarde de / dev / mtdblock0, vous devriez avoir uboot. de toute façon voici ma copie.
> 
> https://ufile.io/rsb6u7fa

Hi can I upload it with the CH341 ? Thanks

klym-tatarnikov, 2020-06-08:
> Hi can I upload it with the CH341 ? Thanks

yes that's definitely what I performed. Just append MTD0..MTD5 into a single file and flashed it with CH341A. Unfortunately it was necessary to completely disassemble the camera and desolder flash chip.

Cseb62, 2020-06-08:
> > Salut, puis-je le télécharger avec le CH341? Merci
> 
> oui c'est définitivement ce que j'ai fait. Ajoutez simplement MTD0..MTD5 dans un seul fichier et flashez-le avec CH341A. Malheureusement, il a fallu démonter complètement l'appareil photo et la puce flash à dessouder.

Sorry but i'm novice. How to do that ?

klym-tatarnikov, 2020-06-09:
> Sorry I'm novice, how to do that :"Just append MTD0..MTD5 into a single file" ?
on windows you can just use command line command "copy":
copy mtd0.bin + mtd1.bin + mtd2.bin + mtd3.bin + mtd4.bin + mtd5.bin  all_together.bin
resulted file should be exactly 16Mb (16777216 bytes)

Cseb62, 2020-06-15:
some one could send me the complete bin? MTD0..MTD5

galichvitaliy, 2021-02-08:
> Hi.
> I've tried to use 'manual' update of IPC016 with new version of FW but something went wrong and now I've bricked the device... It looks U-boot was overwritten.
> 
> Just realized that MTD0 contains u-boot but it's just archived.
> Finally concatenated all MTD* and flashed the chip (with desoldering).

It there any chance you still have .bin for IPC016? I dumped mine with SOIC clip and patched as described, but probably  something went wrong while making backup...

galichvitaliy, 2021-02-08:
> if you did backup of /dev/mtdblock0 you should have uboot. anyway here is my copy.
> 
> https://ufile.io/rsb6u7fa

Do you still have this file? Link is expired, and i completly briked my IPC016. Thank you!

Cseb62, 2021-02-08:
hi, I thinks it's this filesincerely

https://ufile.io/rsb6u7fa

galichvitaliy, 2021-02-08:
> I thinks it's this file https://ufile.io/4hsg0qwh

Thank you for uboot.bin. If anyone can help to figure out, if there are mtd4 and mtd5 in tf_recovery.img?
I tryied to split it like that:

`dd if=tf_recovery.img of=firmware/kernel.bin bs=1 count=2162688
dd if=tf_recovery.img of=firmware/rootfs.bin bs=1 count=7667712 skip=2162688
dd if=tf_recovery.img of=firmware/data.bin bs=1 count=6488064 skip=9830400
dd if=tf_recovery.img of=firmware/config.bin bs=1 count=65536 skip=16318464
dd if=tf_recovery.img of=firmware/factory.bin bs=1 count=65536 skip=16384000`

But there is only 80 bytes left after data.bin, so i don't have mtd4 and mtd5.

galichvitaliy, 2021-02-08:
If somone have telnet or uart access to CMSXJ16A camera, please try to dump missing mtd4, mtd5:

dd if=/dev/mtdblock4 of=/mnt/sdcard/mtdblock4.bin
dd if=/dev/mtdblock5 of=/mnt/sdcard/mtdblock5.bin

rezmus, 2021-02-08:
mtd4 is current camera config and mtd5 factory data with p2p/mijia cloud credentials. you should get them from your dump (at least mtd5), they are not part of recovery img.

SigmaPic, 2021-02-14:
Can some explain how has been found the S49factory exploit ?
And where starting if we want to contribute in order ot find another exploit ?

Haldyz, 2021-03-01:
Sorry for little OT. I have CMSXJ16A with latest fw 16.4.0.9_0405. I wanted to use modded Mi Home app. from vevs, but I have found its not working with latest fw. I want to downgrade to 16.4.0.9_0213 where I have confirmed its working fine and which is available here: http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc016/firmware/IPC016_16.4.0.9_0213.zip (same as other original versions - ie IPC016_16.3.4.5_0081.zip or IPC016_16.3.5.8_0207.zip). I just want to be sure, that this downgrade will not brick my camera. Can anybody confirm this please?

Thank you

jffz, 2021-04-04:
I ended up here searching a hack for "CMSXJ16A", from what I understand you guys achieved to hack it bnut I'm not sure to get the full process to do so. Is the original tf_recovery.img of this repo compatible with it ?

ivan-leschinsky, 2021-08-04:
@jffz any news on the "CMSXJ16A" camera? 
@Haldyz How modified Mi Home app helped you with hack? What did you use?

Haldyz, 2022-01-20:
@ivan-leschinsky it doesn't help me with any hack. As I now, there is no working hack available for CMSXJ16A.
Older fw just allowed me to use it with molded Mi Home app.

leopck, 2022-04-17:
any update on this camera? it doesn't look like there's any update anywhere else on the internet in regards to this camera

KuChar91, 2023-01-15:
Any update on that?

evilmumi, 2023-11-25:
any news here for ipc016?