telmomarques/xiaomi-360-1080p-hacks #8: Dump of SPI flash

jeffreyantony, 2019-02-28:
Is it possible for you to dump the spi flash ?

telmomarques, 2019-03-02:
To be completely honest I wouldn't know where to start...

I currently have access to telnet and U-boot, so if you can provide some pointers on how to do it, I wouldn't mind giving it a try.

jeffreyantony, 2019-03-02:
You need some soldering skills. Try this if you can
https://jcjc-dev.com/2016/06/08/reversing-huawei-4-dumping-flash/

jeffreyantony, 2019-03-02:
Also give a try to read the mtd blocks from the linux itself.
https://stackoverflow.com/questions/13055748/reading-nand-flash-and-writting-it-to-another-system
https://hub.packtpub.com/extracting-data-physically-dd/

telmomarques, 2019-03-02:
Dump of mtd devices on the link below. I'm going to see if I can dump SPI flash over u-boot.

https://drive.google.com/file/d/1FQuHE-6WGTtogbVjOONxt9jtOquyVEN6/view?usp=sharing

jeffreyantony, 2019-03-02:
Is it possible to post the output of the below command. Wanted to understand what blocks were doing

`cat /proc/mtd `

telmomarques, 2019-03-02:
Sure:

```
# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00050000 00010000 "BOOT"
mtd1: 00210000 00010000 "KERNEL"
mtd2: 00750000 00010000 "ROOTFS"
mtd3: 00630000 00010000 "DATA"
mtd4: 00010000 00010000 "CONFIG"
mtd5: 00010000 00010000 "FACTORY"
```

telmomarques, 2019-03-02:
Inside uboot I have this tool
```
MStar # sf
sf - SPI flash sub-system

Usage:
sf probe [[bus:]cs] [hz] [mode] - init flash device on given SPI bus
                                  and chip select
sf read addr offset len - read `len' bytes starting at
                                  `offset' to memory at `addr'
sf write addr offset len        - write `len' bytes from memory
                                  at `addr' to flash at `offset'
sf erase offset [+]len          - erase `len' bytes from `offset'
                                  `+len' round up `len' to block size
sf update addr offset len       - erase and write `len' bytes from memory
                                  at `addr' to flash at `offset'
```

I also have `md` that I can use to dump memory contents:

```
MStar # md 0x0 10
00000000: ea000006 ea000d7c ea000d7c ea000d82    ....|...|.......
00000010: ea000d88 eafffffe ea000d8d ea000d93    ................
00000020: e59f7008 e3a06000 e5876000 ea000000    .p...`...`......
00000030: 1f006010 e59f7008 e3a06000 e5876000    .`...p...`...`..
```

I also know that the firmware lives at 0x50000, and it's read to 0x21000000 before booting.
```
sf probe 0;sf read 0x21000000 0x50000 0x260000;bootm 0x21000000
```

Still haven't figured out to where (physically) these addresses map to. The SPI flash is a Winbond 25q128jvsq which is supposedly 128MiB, but address 0x21000000 should be ~553MiB into the memory?

Do you have any thoughts on how to use this information to dump the SPI flash?

jeffreyantony, 2019-03-02:
My plan was to dump the complete flash and see using a disassembler. There we should be able to see the part where the hash verification is done which we should some how patch so that all the images will be accepted by uboot.

The dumps that you provided before doesn't seem to have the string "libsodium" or "hash". Thats why I requested for the spi dump. Maybe there we should be able to see these strings.

Also I found some tutorials for sf and md here+
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842223/U-boot
http://www.denx.de/wiki/publish/DULG/DULG-tqm8xxl.html#Section_5.9.2.5.

jeffreyantony, 2019-03-03:
https://felipe.astroza.cl/hacking-hi3518-based-ip-camera/

jeffreyantony, 2019-03-09:
Is it possible to dump the complete SPI flash using sf command ?

telmomarques, 2019-03-11:
It should be possible using the `sf` and `md` commands, I just don't know the correct memory addresses to dump...

jeffreyantony, 2019-03-12:
Why not dump till 0x7A12000 (128Mbit) ?

telmomarques, 2019-03-14:
Not so good news: I bricked the camera that I dismantled.

I Inadvertently overwritten u-boot while trying to flash an image directly from the SDCard, on the camera I had opened.

I'm currently working on a second (still intact!) camera. Eventually I have plans to unbrick the other camera, but in the meanwhile I have no means to dump the SPI.

The silver lining to this whole thing is that un-bricking the first camera will - most likely - mean I'll have to solder an ethernet port to the PCB, so **maybe** I'll be able to dump it much faster through tftpd.

vladimirovsan, 2019-04-01:
You did not restore the first camera?

telmomarques, 2019-04-01:
Still haven't restored the camera, been more interested in the RTSP server. At this point I'm not sure if I'll ever will do it :\

rogorman, 2019-04-03:
@telmomarques how do you have access to u-boot - opened camera?

telmomarques, 2019-04-03:
@rogorman yes.
You can check the details in this comment: https://github.com/ghoost82/mijia-720p-hack/issues/10#issuecomment-470599573

karasu, 2019-04-17:
>I've Inadvertently overwritten u-boot while trying to flash an image directly from the SDCard.

Hi, I've just done this myself... I've overwritten /dev/mtdblock0 by mistake. Obviously, camera doesn't boot anymore. Is there any way to reflash it? I've got access to the rx and tx serial ports, but I do not know if there is a way to set the chip in "programming mode" so I can send the u-boot code directly. Is this even possible? @telmomarques it seems you had something in mind to solve this... can you share it?

Thanks a lot!

telmomarques, 2019-04-17:
Hi @karasu, unfortunately I don't have a solution. My bricked camera is still bricked.

The only solution I explored was to solder an ethernet port to the TP, TN, RP and RN contacts, but that wasn't successful (connection was not detected on the computer). I was, most likely, something wrong. So that might be something you want to also try.

If you find anything, do please let me know!

karasu, 2019-04-17:
@telmomarques thanks for your quick answer! I will have a look at the SDK. I'm not sure if I will get any results as I'm more a software guy. I'll keep you posted, though.
Cheers!

fifteenhex, 2019-05-02:
I'm not sure what flash verification is used in this case but the bootrom in the chip drives the "AESDMA" crypto engine to verify the first IPL partition in the SPI flash if there is a key burned into the OTP in the chip itself.
To verify this stuff set the serial baud rate to 38400 and do a power cycle. The bootrom outputs it's stuff at that baud rate before the IPL changes it to 115200.

crckmc, 2019-06-23:
@karasu you can unbrick the camera with an external SPI flasher I'm using an ESP82 for this task. Let me know if you need help

@jeffreyantony just fyi if you havent done it yet: I was successful in patching out the hashing part of uboot so it will flash any image without verification. The downside is that it will now try to flash the recovery image even if it is not present on the sd so you have to replace uboot again once the custom firmware is running

Feel free to joing our Telegram group for discussion and information sharing:
https://t.me/mijav4RTSP

fifteenhex, 2019-06-23:
@crckmc 

I have a port of u-boot for the msc313 if you want to hack it up to boot the vendor kernel instead of trying to hack up the vendor u-boot.

https://github.com/fifteenhex/u-boot/tree/msc313

crckmc, 2019-06-23:
@fifteenhex Yeah I saw (and cloned) your repos thanks for that but it was easier for me to just patch out one if comparison then compile the uboot from scratch and to add the automatic image loading.

elektorpippo, 2019-11-09:
@crckmc can you explain me how unbrick the camera? I've an esp32 to use as SPI programmer. I've already opened the camera to get the serial link but i cannot get the terminal and now  i 've solid yellow led :-( . Thanks.

crckmc, 2019-11-11:
> @crckmc can you explain me how unbrick the camera? I've an esp32 to use as SPI programmer. I've already opened the camera to get the serial link but i cannot get the terminal and now i 've solid yellow led :-( . Thanks.

In theory yes but I only did it with a ESP8266. Have you joined the telegram group yet?

elektorpippo, 2019-11-12:
Yes, I've already joined to the Telegram group. 
Esp8266 and esp32 are very similar, i think it possible only changing the pin configuration.
Thanks

elektorpippo, 2019-11-12:
@crckmc Thanks, i looked for spiff in the group and now all it's clear. Asap i will try your solution!

cosmicDustOfLightLength, 2024-01-07:
> Dump of mtd devices on the link below. I'm going to see if I can dump SPI flash over u-boot.
> 
> https://drive.google.com/file/d/1FQuHE-6WGTtogbVjOONxt9jtOquyVEN6/view?usp=sharing

Link is down, can you reupload?