telmomarques/xiaomi-360-1080p-hacks #9: FW downgrade not possible

seeers, 2019-03-08:
unfortunately i cannot downgrade from FW 3.4.2_0077 to 3.4.2_0062
the file tf_recovery.img  will be renamed to tf_recovery.img.bak after power on.

telmomarques, 2019-03-08:
That's part of the normal flashing procedure. After the camera finishes flashing, the file is renamed "tf_recovery.img.bak"

Please confirm that the downgrade was successful.

seeers, 2019-03-09:
I just tried it again. All files deleted from the SD card, just an empty log folder and the file tf_recovery.img in root. The yellow LED lights for about 15 seconds, then the camera pans back and forth and glows green.
In the Mi Home app I checked for new firmware: Firmware is current: 3.4.2_0077
On the SD card, the file was renamed to tf_recovery.img.bak.

telmomarques, 2019-03-09:
Is there any log file in the SD Card after the procedure? Is yes, please share it.
Also, please confirm the model of your camera. You stated the led glows green but MJSXJ02CM should glow blue.

seeers, 2019-03-09:
Unfortunately there is no logfile after the update start, only at the second start there is one. Of course, in the led I meant blue and not green :) It is also the model MJSXJ02CM. I dismantled the camera the CPU is an ARM MSC313E, MediaTek MT7601U Wifi IC, an unknown IC "MS41929" (presumably the stepper driver) and a W25Q128FV (128M-bit) serial flash memory

telmomarques, 2019-03-09:
Since you've opened the camera, any chance you have a USB FTDI adapter? That way you can get a serial monitor and see what's happening during the flashing procedure.

telmomarques, 2019-03-11:
@seeers any news?

seeers, 2019-03-11:
[image]
Not until now. I am not sure if the RX0 / TX0 are on the board of the CPU or if it is DI DO from the flash memory. I'm trying to find out this evening.

telmomarques, 2019-03-11:
The RX0 and TX0 (marked red) are the right pads. Just connects those to the USB adapter and power on the camera.

seeers, 2019-03-11:
I got a boot log :) 
...
`RTMP_TimerListAdd: add timer obj c4a86ea0!
RTMP_TimerListAdd: add timer obj c4a86ba8!
Start detecting tf_recovery.img
/mnt/sdcard/tf_recovery.img exist, will rename RTMP_TimerListAdd: add timer obj c4a86bdc!
it to /mnt/sdcard/tf_recovery.img.bak
RTMP_TimerListAdd: add timer obj c4a86c10!
RTMP_TimerListAdd: add timer obj c4b081fc!
Starting netcheck...`
....
[output_2019-03-11_19-32-20.log](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/2953729/output_2019-03-11_19-32-20.log)

telmomarques, 2019-03-11:
Weird, I can see you have "tf_recpvery.img" in the SDCard because of this:

```
Start detecting tf_recovery.img
/mnt/sdcard/tf_recovery.img exist, will rename RTMP_TimerListAdd: add timer obj c4a86bdc!
it to /mnt/sdcard/tf_recovery.img.bak
```

But this is missing from the beginning of your bootlog:

```
read file start
reading tf_recovery.img
read len = 0, actlen = 16318544
data check start
Verifying singature using libsodium
Hashing 1048576 bytes, 0 %
Hashing 1048576 bytes, 6 %
(...)
```

Are you sure the image file is not corrupt?

seeers, 2019-03-11:
I tried two different image files, one from here and one from https://en.miui.com/thread-3547398-1-1.html
I'll try another SD card.

telmomarques, 2019-03-11:
Yes, they should be the same, that's where I got it too.
Is the SDCard formatted as FAT32?

seeers, 2019-03-11:
it could have been so easy ... other sd card and it works ....

telmomarques, 2019-03-11:
It's more fun when thing's aren't easy! Glad everything worked out.

closed 2019-03-11 by telmomarques

pedrosimao, 2023-06-15:
@seeers  I am also facing the same issue. I have tried a 32gb card and a 64gb one. I had no luck with any of those.
Can you better describe the card you have used with success and how you formatted it?

tomec7, 2024-01-28:
facing the same. tried 4 different cards.no luck. formatted by camera to FAT32. formatted by PC to FAT32...no luck. can we have details about SD card which worked? thank you

sole-sohail, 2024-12-27:
> Thanks ... I flash my camera ... i access to telnet but i want now to get rtps stream ... How i can ? In VLC rtps://ip don't work ...

Can you please share firmware that supports rtps stream?