telmomarques/xiaomi-360-1080p-hacks #10: How to control camera via telnet?

funivan, 2019-05-01:
Thanks for the hack. All works fine.  I downgrade the firmware and install this hack. Then successfully login via telnet.

Here is my question: how can I turn the camera to the right or left from the telnet? Or execute other commands, for example, enable sleep mode. Thanks.

telmomarques, 2019-05-02:
Hi, unfortunately I didn't have time to look at that, yet.
There might be some way of accessing the GPIO directly, not sure.

CleverNamesTaken, 2019-06-04:
I want to spend some more time looking at the GPIO, but uh... you can turn on and off the LED by controlling the content of /sys/class/gpio/gpio78/value
OFF
echo 0 > /sys/class/gpio/gpio78/value
ON
echo 1 > /sys/class/gpio/gpio78/value

I want to try to take a look at actually controlling the direction of the camera, but I need to find some time to investigate.

telmomarques, 2019-06-04:
Nice work @CleverNamesTaken , thank you for the share!

Jialn, 2019-12-18:
Thanks for the share. Is there any more progress now?

telmomarques, 2020-02-24:
Duplicate of #21.
@Jialn please see that issue, someone has made progress.

closed 2020-02-24 by telmomarques

thewh1teagle, 2020-05-22:
> I want to spend some more time looking at the GPIO, but uh... you can turn on and off the LED by controlling the content of /sys/class/gpio/gpio78/value
> OFF
> echo 0 > /sys/class/gpio/gpio78/value
> ON [...]

Just want to update that you can control every led. 

for yellow led:

ON : 
```
echo 1 > /sys/class/gpio/gpio36
```
OFF:
 ```
echo 0 > /sys/class/gpio/gpio36
```

for blue led:

ON : 
```
echo 1 > /sys/class/gpio/gpio78
```
OFF:
 ```
echo 0 > /sys/class/gpio/gpio78
```