telmomarques/xiaomi-360-1080p-hacks #12: is this hack compatible with CMSXJ16A?

Csontikka, 2019-07-13:
Hi!
is there any chance that this hack is compatible with model CMSXJ16A ?
https://www.aliexpress.com/item/XIAOMI-Mijia-CMSXJ16A-H-265-1080P-IP-Camera-AI-Motion-Detection-Baby-Monitor-360-Pan-tilt/33026548181.html

Thanks!

Csontikka, 2019-07-13:
(my cam has 16.3.4.5_0081 fw now)

telmomarques, 2019-07-13:
Hi,

Can't find any technical details about insides of that camera, but judging from the firmware version it doesn't appear to be the same software...

emersonicus, 2019-08-21:
hey, will your hack work with JTSXJ01CM? 

thanks!

telmomarques, 2019-08-21:
Hi @emersonicus, according to the interwebs that camera uses a different chipset (Ambarella S2LM) so it most likely isn't compatible...

OUARZA, 2020-01-19:
@telmomarques 
Hello i'm looking for the hack for the camera cmsxj16a Do you know where I can find this? thank you

gchrisak, 2020-06-03:
Any news for the CMSXJ16A? 
Its a pretty popular camera due to the features and low cost. Would be great to be able to make it work with Home Assistant.

saarsinai, 2020-07-17:
would love to see support for CMSXJ16A also

javierhurtado, 2020-08-17:
I would like too :)

rezmus, 2020-08-19:
search in issues how to prepare firmware for this model so you get root access. after that you can run rtsp.

javierhurtado, 2020-08-19:
> search in issues how to prepare firmware for this model so you get root access. after that you can run rtsp.

Hello, I would like to, but I dont have knowledge to not brick it :(

teras, 2020-10-01:
> search in issues how to prepare firmware for this model so you get root access. after that you can run rtsp.

Thanks for the tip. You mean [this issue?](https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/18)

javierhurtado, 2020-10-23:
hello, anybody knows how to hack CMSXJ16A?

ElPolloDiabloGDC, 2020-11-27:
Hi, 

i open the CMSXJ16A to get some information. Maybe this data will help for a hack.
If you need other information, just ask.

**Serial Output:**
```
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
SF: 2490368 bytes @ 0x50000 Read: OK
##  Booting kernel from Legacy Image at 21000000 ...
   Image Name:   MVX2##I3gfb35529KL_LX318####[BR:
   Image Type:   ARM Linux Kernel Image (lzma compressed)
   Data Size:    1724780 Bytes = 1.6 MiB
   Load Address: 20008000
   Entry Point:  20008000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... 
[XZ] !!!reserved 0x21000000 length=0x 1000000 for xz!!
   XZ: uncompressed size=0x36c9e0, ret=7
OK
ERR: Can't find KIMG header and initrd address, 0x00000000
atags:0x20000000

Starting kernel

Booting Linux on physical CPU 0x0
Linux version 3.18.30 (jenkins@vm10-2-192-25.ksc.com) (gcc version 4.9.4 (Buildroot 2017.08-gf8e1e38) ) #1 PREEMPT Fri Nov 16 17:00:31 CST 2018
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=50c53c7d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
early_atags_to_fdt() success
Machine model: INFINITY3 MSC000A-S03A-64M
Reserved memory: created CMA memory pool at 0x22c00000, size 20 MiB
Reserved memory: initialized node cma0, compatible id shared-dma-pool
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: console=ttyS0,115200n8r init=/linuxrc rootfstype=squashfs root=/dev/mtdblock2 rw isp_flag=0x0
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 40592K/65536K available (2155K kernel code, 234K rwdata, 1032K rodata, 84K init, 146K bss, 24944K reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
    vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf800000 - 0xc0000000   (   8 MB)
      .text : 0xc0008000 - 0xc0324ecc   (3188 kB)
      .init : 0xc0325000 - 0xc033a000   (  84 kB)
      .data : 0xc033a000 - 0xc03749e0   ( 235 kB)
       .bss : 0xc03749e0 - 0xc0399578   ( 147 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
```

[image]
[image]
[image]

doopeldedoo, 2020-12-01:
> Hi,
>
> i open the CMSXJ16A to get some information. Maybe this data will help for a hack.
> If you need other information, just ask.
>
> **Serial Output:**
>
> ```
> Flash is detected (0x090F, 0x1C, 0x70, 0x18) [...]

Thanks!
I can also help out with information. Have a camera on hand.

patsch9, 2020-12-05:
Tested some packages of manu_test on sdcard with manu.bin and md5sum.dat inside. I can't se if the md5 check (inside firmware /etc/init.d/s49factory) is positive or negative in the logs. The packages has no impact at the moment. Can anyone help?

rogodra, 2021-01-11:
Any news on this camera?, Thanks

b4zyl, 2021-01-18:
I am also interested in hacking this model. Having it working on LAN only would be bless.

gxcreator, 2021-07-29:
More CMSXJ16A output. There is seems uboot console available if you press enter before it boots kernel.
`IPL gd156225
D-01.
HW Reset
64MB
BIST0_0001-OK
offset:00010000
size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4
MXP found at 0x00020000
  decomp_size=0x0004ad64

U-Boot 2015.01 (Aug 12 2019 - 13:56:26), Build: jenkins-ipc016_revert_tutk-2

Version: I3gfe5f65a
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
gpio[100] is 1
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   MAC Address 00:30:1B:BA:02:DB
Auto-Negotiation...
AN failLink Status Speed:10 Full-duplex:0
Status Error!
mstar_emac
Warning: mstar_emac using MAC address from net device

MStar #
MStar #
MStar # help
?       - alias for 'help'
base    - print or set address offset
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
cmp     - memory compare
cp      - memory copy
crc32   - checksum calculation
dbg     - set debug message level. Default level is INFO
dcache  - enable or disable data cache
debug   - Disable uart rx via PAD_DDCA to use debug tool
dhcp    - boot image via network using DHCP/TFTP protocol
dstar   - script via SD/MMC
eeprom  - EEPROM sub-system
env     - environment handling commands
estar   - script via network
estart  - EMAC start
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
fatread - FAT fatread with FSTART
fatsize - determine a file's size
go      - start application at address 'addr'
gpio    - Config gpio port
help    - print command description/usage
i2c     - I2C sub-system
icache  - enable or disable instruction cache
initDbgLevel- Initial varaible 'dbgLevel'
loop    - infinite loop on address range
macaddr - setup EMAC MAC addr
md      - memory display
mm      - memory modify (auto-incrementing address)
mmc     - MMC sub system
mmcinfo - display MMC info
mssdmmc - Mstar SD/MMC IP Verification System
mstar   - script via TFTP
mw      - memory write (fill)
mxp     - MXP function for Mstar MXP partition
net_upgrade- do net update from the specified file that is in tftpserver

nm      - memory modify (constant address)
ping    - send ICMP ECHO_REQUEST to network host
printenv- print environment variables
reset   - Perform RESET of the CPU
riu     - riu  - riu command

run     - run commands in an environment variable
saveenv - save environment variables to persistent storage
setenv  - set environment variables
sf      - SPI flash sub-system
sfbin   - for uploading sf image to a server(via network using TFTP protocol)
srcfg   - sensor pin and mclk configuration.
tftpboot- boot image via network using TFTP protocol
version - print monitor, compiler and linker version
`

gxcreator, 2021-07-29:
Get ssh running, replace http://192.168.1.224:8000/hacks/ssh-server/bin/dropbear with your url to armv7l dropbear
`
killall hostapd
echo -ne "network={\n    ssid=\"gx-iot\"\n    psk=\"YOUR_PASSWORD\"\n}\n" > /tmp/wpa.conf
wpa_supplicant -B -i wlan0 -c /tmp/wpa.conf -D wext

udhcpc -i wlan0

ping -c 2 8.8.8.8

curl -k http://192.168.1.224:8000/hacks/ssh-server/bin/dropbear --output /tmp/dropbear
cd /tmp
chmod +x dropbear

# cat config.json
echo -ne "{\"users\": [{\"systemUsername\": \"root\", \"username\": \"root\", \"password\": \"\"}]}" > config.json
./dropbear -FREB -r ./host_150601_rsa -r ./host_150601_rsa -r ./host_150601_rsa -C config.json
`

OUARZA, 2021-08-01:
@gxcreator 
Hello,
Could you tell us how to retrieve the feed from the CMSXJ16A camera?
thanks a lot

gxcreator, 2021-08-03:
@OUARZA Check this thread with photos:
https://github.com/roleoroleo/yi-hack-MStar/issues/376

ivan-leschinsky, 2021-08-04:
Have the same CMSXJ16A camera, I think about this about 2 years, if we can actually view this camera rtsp without xiaomi cloud?

OUARZA, 2021-11-14:
Hello @gxcreator ,
were you able to move forward on the subject?
Mat

vovka1981, 2021-12-05:
Hello @gxcreator and I also would be pleased to get it work on CMSXJ16A

yakovte, 2022-04-09:
Me to i withing for hack cmsxj16a

evilmumi, 2023-11-25:
any achivements`?

mgmorpheus77, 2024-02-13:
Someone else is working on this topic?

mgmorpheus77, 2024-02-20:
I can't believe that such wise minds have given up on this topic :/

floxigen, 2024-03-03:
Sad we don't have support for CMSXJ16A