telmomarques/xiaomi-360-1080p-hacks #15: Compatibility with CMSXJ13B

ppisljar, 2019-09-26:
Is this compatible ? 

I tried downloading tf_recovery.bin to sdcard and flashing. The file was renamed but nothing happened. I have yet to try with different SD card. Is there a procedure i could follow to add compatibility in case its not yet working ? 

How similar/different is this to other xiaomi mija hacks (https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks, https://github.com/TheCrypt0/yi-hack-v4, ...) ? Are they all based on the same technique ? and how hard would it be to bring some of their features into this or add support for this camera to their versions ?

thanks!

telmomarques, 2019-09-26:
I'm not finding much information on this camera. Do you happen to know the SoC (processor) brand and model?

ppisljar, 2019-09-27:
no idea, i'll have to take it apart .... is there any guide on doing so ? 

https://www.mishop.si/img/p/4/9/1/9/4919-large_default.jpg

ppisljar, 2019-09-27:
i doubt i'll ever be able to put it back together :)

[image]
[image]
[image]

crckmc, 2019-09-27:
Looks like it is the same SoC Mstar313E.
Do you have access to a soldering iron?

If yes try to get serial access with the pads TX0, RX0 and GND.
You will need a USB TTL converter.

ppisljar, 2019-09-27:
got it connected ... whats the speed ? i tried with 9600 and 115200 and its just garbage

crckmc, 2019-09-27:
It should be 115200 are you sure there is no ground problem?

ppisljar, 2019-09-27:
```
▒▒▒▒▒
IPL gd156225
D-01.
HW Reset
64MB
BIST0_0001-OK
[NOR]
offset:00010000
size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4
MXP found at 0x00020000
  decomp_size=0x0004ad24

U-Boot 2015.01 (Dec 21 2018 - 17:50:28), Build: jenkins-IPC013D_english-133

Version: I3g574ecbf
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
gpio[100] is 1
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   MAC Address 00:30:1B:BA:02:DB
Auto-Negotiation...
AN failLink Status Speed:10 Full-duplex:0
Status Error!
mstar_emac
Warning: mstar_emac using MAC address from net device

read file start
reading tf_recovery.img
** Unable to read file tf_recovery.img **
read len = -1, actlen = 0
read file tf_recovery.img error.
[NetUpgrade] ts_1st=0x10d4
[NetUpgrade] Get Device MacAddr(00:30:1b:ba:02:db)
NetUpgrade Timeout
[NetUpgrade] ts_1st=0x10d4, ts_2nd=0x14c0, delt_ts=0x3ec
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
SF: 2490368 bytes @ 0x50000 Read: OK
##  Booting kernel from Legacy Image at 21000000 ...
   Image Name:   MVX2##I3g574ecbfKL_LX318####[BR:
   Image Type:   ARM Linux Kernel Image (lzma compressed)
   Data Size:    1724656 Bytes = 1.6 MiB
   Load Address: 20008000
   Entry Point:  20008000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ...
[XZ] !!!reserved 0x21000000 length=0x 1000000 for xz!!
   XZ: uncompressed size=0x36c9e0, ret=7
OK
ERR: Can't find KIMG header and initrd address, 0x00000000
atags:0x20000000
[... 634 lines cut ...]
MT7601_ChipSwitchChannel: SwitchChannel#1(RF=15, 1T)
80211> cfg80211_scan_done
[1114][Debug][CFG80211OS_ScanEnd] FlgIsAborted = 0
#####check WPS IE
ConnInfo.bWpsConnection ===> Not WPS IE
80211> Connect bssid 3c:37:86:29:14:07
CNTL - iterate BSS 0 of 4
MT7601_ChipSwitchChannel: SwitchChannel#4(RF=15, 1T)
PeerBeaconAtJoinAction(): HT-CtrlChannel=4, CentralChannel=>4
PeerBeaconAtJoinAction(): Set CentralChannel=4
MT7601_ChipSwitchChannel: SwitchChannel#4(RF=15, 1T)
udhcpc: performing DHCP renew
udhcpc: sending discover
RTMP_TimerListAdd: add timer obj c4bc5ab0!
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Rcv Wcid(1) AddBAReq
Start Seq = 00000002
RTMP_TimerListAdd: add timer obj c4bc7ad0!
udhcpc: sending select for 10.0.0.68
udhcpc: lease of 10.0.0.68 obtained, lease time 86400
deleting routers
adding dns 10.0.0.1
get ip addr :
10.0.0.68
```

ppisljar, 2019-09-27:
it complains about not being able to read the file ....

ppisljar, 2019-09-27:
tried different card ... looking good:
```
IPL gd156225
D-01.
HW Reset
64MB
BIST0_0001-OK
offset:00010000
size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4
MXP found at 0x00020000
  decomp_size=0x0004ad24

U-Boot 2015.01 (Dec 21 2018 - 17:50:28), Build: jenkins-IPC013D_english-133

Version: I3g574ecbf
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
gpio[100] is 1
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   MAC Address 00:30:1B:BA:02:DB
Auto-Negotiation...
AN failLink Status Speed:10 Full-duplex:0
Status Error!
mstar_emac
Warning: mstar_emac using MAC address from net device

read file start
reading tf_recovery.img
read len = 0, actlen = 16318544
data check start
Verifying singature using libsodium
Hashing 1048576 bytes, 0 %
Hashing 1048576 bytes, 6 %
Hashing 1048576 bytes, 12 %
Hashing 1048576 bytes, 19 %
Hashing 1048576 bytes, 25 %
Hashing 1048576 bytes, 32 %
Hashing 1048576 bytes, 38 %
Hashing 1048576 bytes, 44 %
Hashing 1048576 bytes, 51 %
Hashing 1048576 bytes, 57 %
```

ppisljar, 2019-09-27:
yeey flashed.

can you maybe answer my above question about other xiaomi camera hacks ? each one is for different IC ? or do some of  the hacks target the same IC ? how hard would it be to take some of the features and bring it in, or add support for our devices to one of the projects with more features ?

crckmc, 2019-09-27:
@ppisljar as far as I know the yi and dafang cameras have a different chipset/software so I think it would be hard to port the features of the other hacks

telmomarques, 2019-09-29:
Great work guys, glad you got it working!

MarcosAndre, 2020-04-17:
Sorry guys, I have a camera with that module, can you help me to add the Telmo hacks?

MarcosAndre, 2020-04-17:
@ppisljar :)

edugrt, 2020-11-11:
@MarcosAndre, hey, did you manage to flash it?

MarcosAndre, 2020-11-11:
Nop, I prefer to wait for a stable version.

ppisljar, 2020-12-21:
sorry to disappear, update from my side: managed to flash the camera but the firmware doesn't work out of the box and i gave up

edugrt, 2021-01-03:
Hi, @telmomarques. Would you be able to tell us how hard would it be to adapt the firmware to this model version also? I’d like to know if I could help in any way. Thanks!