telmomarques/xiaomi-360-1080p-hacks #17: TelNet?

EscVM, 2019-10-30:
Hi! Thank you for this repository!

I followed the guide and now I'm able to connect to my camera with Telnet! However, what can I do with it? Is it possible in someway to grab the frames or the streaming video?

EscVM, 2019-10-30:
Where can I find the repo? I would really be interested in it! Thx

crckmc, 2019-10-30:
https://github.com/crckmc/mija-framegrabber

EscVM, 2019-10-30:
Instead, what about this project xiaomi-360-1080p-video-stream ? Does it work? In which folder do I have to put ffmpeg and video_direct_memory_access files?

EscVM, 2019-10-30:
> https://github.com/crckmc/mija-framegrabber

How can I use it. I cannot understand how to use it from the guide.

EscVM, 2019-10-30:
Find it! Thanks!