telmomarques/xiaomi-360-1080p-hacks #18: Compatibility with Mijia 1080p MJSXJ05CM ?

Jayah59, 2019-11-13:
Hi !

Is this compatible ?
I tried downloading tf_recovery.bin to sdcard and flashing but it did not work and I worse, I think my camera is bricked ...

LED amber is not blinking, it is permanent on and the camera does not move. 

>.<

crckmc, 2019-11-13:
Looks like you found out it isnt compatible :)

Try to find the recovery image for your camera type.
uboot is not overwritten so should still be working

Jayah59, 2019-11-13:
> Looks like you found out it isnt compatible :)
> 
> Try to find the recovery image for your camera type.
> uboot is not overwritten so should still be working

Yes indeed 😅

I'm trying to find a recovery image but I'm struggling. I opened a post on the Xiaomi forum, I hope someone can help me.

I started to disassemble the camera to connect a serial port. As soon as possible, I will solder the serial port and I will post the result.

Thx

Jayah59, 2019-11-15:
Hello! 
I'm coming back because I disassembled the camera. Now I would like to solder a serial port but I do not know where. 

[image]

Can somebody help me please ?  :D

Jayah59, 2019-11-15:
Maybe we would like the ARM chip  :)

[image]

none815, 2020-01-04:
I received this camera today with the same model number.
[Current firmware can be found here.](https://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc019/firmware/IPC019_3.5.1_0052.zip)

none815, 2020-01-05:
A firmware hack is possible, but requires some tools:
- CH341A flash programmer
- SOIC8 clip and some dupont wires
- Complete device teardown

Basic steps are:
- Create a flash backup:
`flashrom -p ch341a_spi -r backup`

- [Download the firmware](https://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc019/firmware/IPC019_3.5.1_0052.zip) and put it into the same folder.

- Run this script to patch the backup:
```
#!/bin/bash
# extract firmware
mkdir -p files
dd if=tf_recovery.img of=files/kernel.bin bs=1 count=2097152
dd if=tf_recovery.img of=files/rootfs.bin bs=1 count=7733248 skip=2097152
dd if=tf_recovery.img of=files/data.bin bs=1 count=6488064 skip=9830400

# patch jffs2 partition
sudo modprobe mtdblock
sudo modprobe mtdram total_size=6336
sudo dd if=files/data.bin of=/dev/mtdblock0 bs=1
mkdir mount
sudo mount -t jffs2 /dev/mtdblock0 mount
echo '#!/bin/sh' | sudo tee -a mount/bin/log_diag_platform.sh
echo '/mnt/sdcard/override.sh' | sudo tee -a mount/bin/log_diag_platform.sh
sudo chmod 755 mount/bin/log_diag_platform.sh
sudo umount mount
rmdir mount
sudo dd if=/dev/mtdblock0 of=files/data.bin bs=1

# update flash backup
mkdir -p out
cp backup.bin -f out/flash.bin
dd if=files/kernel.bin of=out/flash.bin bs=1 count=2097152 seek=327680
dd if=files/rootfs.bin of=out/flash.bin bs=1 count=7733248 seek=2424832
dd if=files/data.bin of=out/flash.bin bs=1 count=6488064 seek=10158080
dd if=/dev/zero bs=1 count=131072 | tr "\000" "\377" >> out/flash.bin
```

- Re-flash the modified backup:
`flashrom -p ch341a_spi -w out/flash.bin`

none815, 2020-01-05:
The flash chip is located right beside the camera sensor, it is necessary to partially remove the lens:

[image]
[image]

It might occur that the SOIC clip will also power up the camera, this will interfere the flash programming. One solution is to cut HOLD (pin7) and VCC (pin8) to reduce power delivery, this can be done with some dupont wires:

[image]

phamthanhtri, 2020-01-14:
@Jayah59  Have you successfully hacked  MJSXJ05CM yet ? I also want to hack but don't know what to do :)

ahmetikbal, 2020-01-15:
@Jayah59 @phamthanhtri @none815  firmware update didn't complete and my MJSXJ05CM bricked. I want to hack with tf_recovery.bin file. If you have for 05CM, can you share?

phamthanhtri, 2020-01-16:
@ahmetikbal   you can use this file to reset camera into stable version (https://drive.google.com/open?id=1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj). I still haven't hacked yet

gurkburk76, 2020-01-16:
RIght, so i have version mjsxj05cm, could the firmware posted by @phamthanhtri be hacked so telnet can be enabled? at least that would be a start towards getting rtsp working i guess :)

DaeMonSx, 2020-02-17:
still nothing at MJSXJ05CM hack?
It seems nowadays thise are the cameras found in shops.

Fruityski, 2020-03-13:
I have managed to un-brick my MJSXJ05CM  with https://drive.google.com/file/d/1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj/view

Any update on the hack?

KhArtNJava, 2020-05-24:
+ 1 to those who are waiting for MJSXJ05CM support.

UndeRus, 2020-05-24:
> The flash chip is located right beside the camera sensor, it is necessary to partially remove the lens:
>
> [image]
> [image]
>
> It might occur that the SOIC clip will also power up the camera, this will [...]

How did you flash this chip with cutted 7 and 8 wires? My programmer didn't detect chip when these wires are cutted

kathgironpe, 2020-06-19:
Hello...

I have a bricked camera and I need to flash the default firmware.
Because I have iPhone & other mobile phone repair kits, it was not a challenge to tear down the camera. Well, of course, it was challenging for the bottom part.

When I run this command:

```sh
 flashrom -p ch341a_spi -r backup.bin
```

I get:

```sh
flashrom v1.2 on Darwin 19.6.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Couldn't open device 1a86:5512.
Error: Programmer initialization failed.
```

What I use:

25 series set of CH341A flash programmer w/ the clip. No soldering done.

The command `lsusb` does recognize the device plugged:

```sh
Bus 000 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

lucperreau, 2020-07-05:
I am also waiting for MJSXJ05CM flash. Don't want to open the camera up. Thanks :)

KhArtNJava, 2020-07-08:
> A firmware hack is possible, but requires some tools:
> 
> * CH341A flash programmer
> * SOIC8 clip and some dupont wires
> * Complete device teardown
> 
> Basic steps are:
> 
> * Create a flash backup:
>   `flashrom -p ch341a_spi -r backup.bin`
> * [Download the firmware](https://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc019/firmware/IPC019_3.5.1_0052.zip) and put it into the same folder.
> * Run this script to patch the backup:
> 
> ```
> #!/bin/bash
> # extract firmware
> mkdir -p files
> dd if=tf_recovery.img of=files/kernel.bin bs=1 count=2097152
> dd if=tf_recovery.img of=files/rootfs.bin bs=1 count=7733248 skip=2097152
> dd if=tf_recovery.img of=files/data.bin bs=1 count=6488064 skip=9830400
> dd if=backup.bin of=files/vendor.bin bs=1 count=131072 skip=16646144
> 
> # patch jffs2 partition
> sudo modprobe mtdblock
> sudo modprobe mtdram total_size=6336
> sudo dd if=files/data.bin of=/dev/mtdblock0 bs=1
> mkdir mount
> sudo mount -t jffs2 /dev/mtdblock0 mount
> echo '#!/bin/sh' | sudo tee -a mount/bin/log_diag_platform.sh
> echo '/mnt/sdcard/override.sh' | sudo tee -a mount/bin/log_diag_platform.sh
> sudo chmod 755 mount/bin/log_diag_platform.sh
> sudo umount mount
> rmdir mount
> sudo dd if=/dev/mtdblock0 of=files/data.bin bs=1
> 
> # update flash backup
> mkdir -p out
> cp backup.bin -f out/flash.bin
> dd if=files/kernel.bin of=out/flash.bin bs=1 count=2097152 seek=327680
> dd if=files/rootfs.bin of=out/flash.bin bs=1 count=7733248 seek=2424832
> dd if=files/data.bin of=out/flash.bin bs=1 count=6488064 seek=10158080
> dd if=files/vendor.bin of=out/flash.bin bs=1 count=131072 seek=16646144
> ```
> 
> * Re-flash the modified backup:
>   `flashrom -p ch341a_spi -w out/flash.bin`
> * Prepare the sdcard script:
> 
> > /sdcard/override.sh
> 
> ```
> #!/bin/sh
> main() {
>   # start telnet
>   /mnt/sdcard/busybox telnetd
> }
> 
> if [ ! -f /tmp/.override ]; then
>  touch /tmp/.override
>  main
> fi
> ```
> 
> * Press the reset button (briefly) on the camera to launch the override script.

none815, can you tell us please, what flash chip installed in MJSXJ05CM? As I understand, it's SPI flash. But what is series/number of the chip? What's flash size in that chip?

gusi1994, 2020-07-16:
@none815 I did flash the chip like you suggested but it didn‘t work properly. Any suggestion what the steps are necessary after resetting the camera?

gusi1994, 2020-07-16:
@KhArtNJava 
> none815, can you tell us please, what flash chip installed in MJSXJ05CM? As I understand, it's SPI flash. But what is series/number of the chip? What's flash size in that chip?

I did a little bit of research and the Chip seems to be the cFeon Q32B-104HIP, 32Mbit SPI Serial Flash, SOIC-8 or also many times referenced as EN25Q32B.

https://www.kean.com.au/oshw/WR703N/teardown/EN25Q32B%2032Mbit%20SPI%20Flash.pdf

Does this help?

slock83, 2020-08-08:
Hello everyone,

So I tried @none815 method and it worked well.
The flash chip, in my case, was an EN25QH128.

I did a few modification though, instead of having the script launched by the diagnosis launcher, I modified wifi_start, which allows me to run my script on boot, instead of on reset.

I'm still working on getting the current hacks to run, runsvdir doesn't seem to work yet ...

Thanks !

fumanchi, 2020-08-12:
Hi slock82,

I am stuck with runsvdir as well...
the file itself seams to be there but i always get:
"/mnt/sdcard/manu_test/configure_services.sh: line 50: runsvdir: not found"
As if the file wouldn't be there...
I found the file has been compiled dynamically...
It directly depends on:
Shared library: [libc.so.6]
Shared library: [ld-linux-armhf.so.3]
which seams to be fine...

As we are using a newer firmware they might have blocked the system from running binaries from sdcard. what do you think?
Have you made any progress?

puuhderbaer, 2020-08-17:
Hello everybody,

today my programmer arrived and I managed to successfully flash the spi-flash with the by none815 described method. Cam is powering up correctly and re-setup went fine.

But now I'm stuck with the mentioned override.sh script. telnet is to be run by busybox `/mnt/sdcard/busybox telnetd` , right? Can anybody please tell me what I'm missing? Where to get busybox from?

Thanks in advance!

twosky2000, 2020-08-17:
New busybox would contain runit (runsv/dir).
@puuhderbaer busybox can be downloaded here: https://busybox.net/downloads/binaries/1.31.0-defconfig-multiarch-musl/busybox-armv7l
Also I got runit with busybox working.
`/mnt/sdcard/busybox --list | awk '/runsv|chpst|svlog|^sv$/' | xargs -I{} ln -sv /mnt/sdcard/busybox /mnt/data/bin/{}`
I'm not sure how to make it permanent jet.

Also my MJSXJ05CM uses Camera model 'ipc019' not ipc009, we would have to recompile the mija-framegrabber.

puuhderbaer, 2020-08-18:
Thanks a lot @twosky2000! That did the trick. Telnet is up and running. But there is not yet any progress getting rtsp to work on 'ipc019', right?

barusbarus, 2020-08-18:
Witam serdecznie.
Szukam pomocy w sprawie uruchomienia zablokowanej kamery Mija Smart 2k MJSXJ09CM. 
Po wielu próbach z różnymi wersjami programówi "tf_recovery" w końcu udało mi się ją odblokować po nieudanej aktualizacji.
Kamera ożyła, miga żółta dioda, wydaje komunikaty dzwiękowe, wykonuje obrót i można ją resetować ale  nie mogę ją skonfigurować/ połączyć z serwerem w chinach, tak jak było to możliwe przed feralną aktualizacją.
Po zeskanowaniu kodu z dolnej części kamery MiHome rozpoznaje kamerę, w następnym kroku próbuje łączyć się z dedykowaną siecią wifi, po pewnym czasie wydaje komunikat aby wybrać inną sieć  "chuangmi-camera-ip029a_miapAF60"
po nawiązaniu połączenia z urządzeniem wyświetla komunikat aby wybrać inną sieć w celu wysłania wiadomości, niestety nie jest w stanie wysłać tej wiadomości.
[image]
[image]
[image]
[image]

Prawdopodobnie flash nie jest dedykowany do tej kamery.
Czy istnieje jakiś inny sposób na poprawne skonfigurowanie tej kamery, czy muszę wgrać inny flash dedykowany do tej kamery,
czy w ogóle da się uratować tą kamerę?
Bardzo proszę o pomoc.
Z góry dziękuję.

rezmus, 2020-08-18:
@barusbarus create new issue if you have different camera model.

here is recovery firmware for your cam.

https://ufile.io/r2kimbcl

jandy123, 2020-08-29:
Since I do not have the HW tools required to flash the EEPROM, I was wondering what would happen if I just patch the jffs2 partition as described by @none815 and try to flash via OTA update ? Would this work ? HAs anyone already tried this ? 

Thanks !

fumanchi, 2020-09-07:
Has anybody experience in cross compiling for that architecture? I do have obtained the SDK but do not have any idea which toolchain to use and so on...

fumanchi, 2020-09-07:
> Well, could you share the SDK ? I managed to cross-compile mosquitto to signal motion detect events to hassio. It's a nuisance, though. I simply used arm-linux-gnueabihf on a standard ubuntu 18.04 install. Some libraries had to be pulled from the camera to get it link.

The SDK is freely available... 
https://dl.openipc.org/SDK/MStar

jandy123, 2020-09-07:
Thanks ! I see the toolchain sources there.

fumanchi, 2020-09-07:
I do so as well.. but i do not know what i actually need to compile e.g. the mija-framegrabber...
I do work on an ubuntu 16.04 LTS at the moment...
A toolchain is a set of compilers (e.g. cc, cxx),and archivers (e.g. ar).. but there are no such linux binaries (elf binaries for amd64)... Have I missed something?

jandy123, 2020-09-07:
Could you please share the SDK and toolchain using e.g. google drive ?
I cannot download them for free from the location you indicated. Then,
I could have a look at the toolchain.

jandy123, 2020-09-07:
Thanks for the link, but is this the same with the one from
https://dl.openipc.org/SDK/MStar ? File naming suggests otherwise.

fumanchi, 2020-09-07:
> Thanks for the link, but is this the same with the one from https://dl.openipc.org/SDK/MStar ? File naming suggests otherwise.
> […](#)

This is all I have :(

jandy123, 2020-09-07:
Anyways, back to your question, I suppose you mean compile this: https://github.com/crckmc/mija-framegrabber

I understand all steps, but the problem is that I do not have on MJSXJ05CM. 
libpthread-2.25
libc-2.25

For the rest it should all be doable with the toolchain from https://dl.openipc.org/SDK/MStar and the SDK.

telmomarques, 2020-09-07:
Check https://github.com/telmomarques/mija-framegrabber/
The makefile has targets for ipc019 (MJSXJ05CM), the `LIBS` variable is the shared libraries you need.

fumanchi, 2020-09-07:
> Anyways, back to your question, I suppose you mean compile this: https://github.com/crckmc/mija-framegrabber
> 
> I understand all steps, but the problem is that I do not have on MJSXJ05CM.
> libpthread-2.25
> libc-2.25
You mean you do not have the headers aka development files? Shouldn't the api (at least as far as it might be used in the frame grabber sources) compatible to other gnu platforms (e.g. linux?)

> For the rest it should all be doable with the toolchain from https://dl.openipc.org/SDK/MStar and the SDK.
 Yes.. I think we should try to get the toolchain anyhow... the SDK is not helping a lot :(

fumanchi, 2020-09-07:
> Check https://github.com/telmomarques/mija-framegrabber/
> The makefile has a target for ipc019 (MJSXJ05CM), the `LIBS` variable is the shared libraries you need.
So we need the shared libraries at compile time :( are they not available on the cam?
At least the libpthread.so and librt.so are part of the SDK/arm-linux-gnueabihf)...

telmomarques, 2020-09-07:
Yes, they are available on the cam 🙂
Notice on the makefile that the libs for MJSXJ05CM are different, you don't need libpthread or librt for target ipc19..

telmomarques, 2020-09-07:
I noticed I still have framegrabber compiled locally for ip019. So if you want to try it out I uploaded a new release:
https://github.com/telmomarques/mija-framegrabber/releases/tag/temp

jandy123, 2020-09-07:
Well, I've just tried. It compiles and links fine (using libs pulled from the camera),  but at runtime it still has a dependency:

framegrabber: can't load library 'libc.so.6'

Any idea ?

telmomarques, 2020-09-07:
@jandy123, IPC019 does not use libc (it uses uClibc), so I think you're using the wrong binary. Check my previous message for a compiled binary of framegrabber for IPC019.

jandy123, 2020-09-07:
@telmomarques:
Yes, I've seen the libs used from the camera and copied them from there. I also had to modify the makefile for the toolchain path. 

I'm sure I compile just fine and for target IPC019. I still get the libc dependency, which I do not understand... The only explanation may be that you use a different toolchain. Could you please share it ?

telmomarques, 2020-09-07:
I used [buildroot](https://buildroot.org/) to build a cross-compile toolchain for armhf with uClibc.
Here's my buildroot `.config` if you want to try it: [config.zip](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/5182411/config.zip)

jandy123, 2020-09-07:
@telmomarques: Thanks for the config file. I'll first test the binary you provided and see what happens. as a last resort I'll compile buidlroot.

Are you aware of a smarter way of detecting motion events on the camera ? Right now I have a stupid shell script watching /tmp/cloud. Whenever *.mp4 appear in that folder, I signal to my MQTT broker using mosquitto_pub. 

Thanks again !

telmomarques, 2020-09-07:
Regarding motion detection, at this time no, sorry :\

jandy123, 2020-09-07:
Well, the framegrabber you provided definitely works !

fumanchi, 2020-09-07:
@jandy123 So, the newly provided frame grabber (for IPC019) works as you say... am I right assuming the whole hack here works with the new frame grabber deployed on the sdcard?

jandy123, 2020-09-07:
Well, I did not properly test it. I've just started, saw it running and stopped it... But yes, all hacks run from an sdcard. Please read carefully the installation instructions.

fumanchi, 2020-09-07:
@jandy123 I did :) a also managed to hack the cam using a programmer (CH341A) to be able to run arbitrary shell scripts or binaries (e.g. busybox) off my sd card... my quation was if all the "modules" the hack consists off are running on the new revision cams now? So that I just need to exchange the frame grabber binaries and it should work... right?

jandy123, 2020-09-07:
@fumanchi: I see, sorry my bad. I did not test the various modules yet, so I don't know, sorry. I'm still struggling to get a cross-compiler working without a lot of headache ;).

rezmus, 2020-09-07:
no, most binaries for MJSXJ02CM are not compiled against uclibc. however if you run framegrabber and rtsp server (h265) only you will be fine.

jandy123, 2020-09-07:
@telmomarques: I can confirm your buildroot toolchain does the trick. 

Thanks again !

fumanchi, 2020-09-07:
@rezmus Thanks... The retspserver binary file is (or at least seams to be) statically built/linked but does not understand the current arguments (here: "-c ...json" ~> /mnt/data/bin/rtspserver: invalid option -- 'c'
I probably need to compile buildroot and the rtspserver on my own :(

fumanchi, 2020-09-07:
@jandy123 Do you managed to build a working rtspserver binary from current sources? I would be very pleased to get my hands on a working rtspserver :)

rezmus, 2020-09-07:
@fumanchi 

you only need -m -s for pipes. look at alpha.v3.zip\hacks\rtsp-server\meta\service\ for the scripts which starts framegrabber/rtspserver.

fumanchi, 2020-09-07:
@rezmus I see.. but no success... the binaries run/work.. but i neither get a valid rtsp stream nor a webstream...
the named pipes in /mnt/data/etc/framegrabber/ do provide data but i cannot get any connection to the rtsp or web stream...

rezmus, 2020-09-07:
can you see the stream in mi home app? it works for me.

extract alpha v3 hacks, replace framegrabber with uclibc version.

# cd /mnt/sdcard/hacks/rtsp-server/meta/service/
# ./framegrabber-mainstream &

open 2nd telnet window.

# cd /mnt/sdcard/hacks/rtsp-server/meta/service/
# ./rtsp-server &

connect with vlc to rtsp://cameraip:8554/mainstream

2879597772, 2020-09-08:
> @KhArtNJava
>
>
> I did a little bit of research and the Chip seems to be the cFeon Q32B-104HIP, 32Mbit SPI Serial Flash, SOIC-8 or also many times referenced as EN25Q32B. [...]

Does anybody know how this was obtained? I used the Binwalk tool and didn't understand how it worked

[192 more replies]