telmomarques/xiaomi-360-1080p-hacks #19: Token via Telnet

sicciaman, 2019-11-27:
I'm trying to find Xiaomi 360 1080P camera through Telnet but I can't find where it's located

telmomarques, 2019-11-27:
I don't know if I understood your question correctly, you can't find the IP address of the camera?
You can find the IP in the Mi Home app:
General settings -> More settings -> Network info

sicciaman, 2019-11-27:
No, I have the IP of the camera, but to communicate using miio-python library, it's necessary also a Token

telmomarques, 2019-11-27:
I'm not sure this version of the camera is supported by miio-python.

I can see in the documentation that there are multiple ways of obtaining tokens, check the docs and try the documented methods.

sicciaman, 2019-11-27:
I've already try some of those methods, but with this Camera seems not working. So in another firmware Github page I've found that it's possible get the token directly from Telnet. I'm searching between all directories and files but I didn't found nothing.

telmomarques, 2019-11-27:
Seems like this camera model is not supported by that library... You can maybe try reaching out to python-miio repo for help.

sicciaman, 2019-11-27:
Ok thanks anyway!

closed 2019-11-27 by telmomarques

sicciaman, 2019-11-27:
Found a way to get Token for Xiaomi Mijia 360 1080p (with your Firmware). Connect via Telnet to the camera, follow the address: /mnt/data/bin and execute command mortoxd that show some info (token is in the field: miio_token