telmomarques/xiaomi-360-1080p-hacks #20: sdcard file structure

pickeld, 2019-12-04:
Hey, i successfully downgraded the camera.
i copied both **folders** (hacks + menu_test) to the root of the sdcard. 
turn the camera on but cant reach the web
what am i missing?
thanks

telmomarques, 2019-12-04:
Hey,

Can you please specify what do you mean by "reach the web"?

pickeld, 2019-12-04:
thanks for the quick replay. im trying to get the camera on the <IP>:80.

pickeld, 2019-12-04:
sorry, just realized i should telnet

dr400, 2020-06-09:
telnet is sending back a "connection refused" ... port is 23

thewh1teagle, 2020-06-09:
@dr400 
I think that telnet is not running in last version. Try ssh instead.

dr400, 2020-06-09:
> @dr400
> I think that telnet is not running in last version. Try ssh instead.

you're right ! ssh is working great but not telnet ...

telmomarques, 2020-06-09:
> > @dr400
> > I think that telnet is not running in last version. Try ssh instead.
> 
> you're right ! ssh is working great but not telnet ...

Telnet was replaced by SSH in the latest version.

closed 2020-06-09 by telmomarques