telmomarques/xiaomi-360-1080p-hacks #21: Any way to control the motor?

Jialn, 2019-12-18:
I have this product, and want to make a tele controlled robot by replacing the cable of original motor with DC motor driver pins on the robot. 
That might require controlling to the motor by a customized service, anyway to do this?

telmomarques, 2019-12-18:
Hi!

You can try asking on the telegram group (you can find the link in another issue), I think someone has posted something about controlling the motor, not 100% sure though.

Sorry for not being able to help more.

Jialn, 2019-12-18:
Thanks for the quick replay! I'll check that.

Jialn, 2019-12-18:
Sorry, any hint to which issue has the link of telegram group? Can not find it.

telmomarques, 2019-12-18:
Sure, here it is: https://t.me/mijav4RTSP

thewh1teagle, 2020-02-24:
Hi 
I created repo for that thing, you can control the motor directly with my code. 
https://github.com/thewh1teagle/xiaomi-1080-360-motor-control

closed 2020-09-18 by Jialn