telmomarques/xiaomi-360-1080p-hacks #22: [new hack] Block Cloud

bread-on-toast, 2020-01-04:
Hi,
I wrote a dirty hack to block the camera to connect to the cloudserver for privacy reasons. Feel free to use:
https://github.com/bread-on-toast/xiaomi-360-1080p-reroute-cloud

Arcrocus, 2020-10-06:
> Hi,
> I wrote a dirty hack to block the camera to connect to the cloudserver for privacy reasons. Feel free to use:
> https://github.com/bread-on-toast/xiaomi-360-1080p-reroute-cloud

Great job! Maybe you can do a hack to make Xiaomi 360 1080p camera send videos from sdcard to another cloud service (for example Google Drive)?

thewh1teagle, 2020-10-06:
Great. 
Is there a chance that there's another network endpoints that you didn't covered?
Maybe we need to capture the traffic to be able to block anything that's related to the original firmware.

bread-on-toast, 2020-10-08:
@thewh1teagle , acctually it might be, however I hadn't noticed them. However, implementing more endpoints would be straight forward, so if you see some let me know.

@Arcrocus the original Hack features a RTSP Server now, so I assume if you route the cameras Port through your local NAT to the outside world, any stream client / capture service could be used.