telmomarques/xiaomi-360-1080p-hacks #25: Can't downgrade firmware

memma89, 2020-04-08:
Hello, i followed your instructions but i was unable to downgrade the firmware. 
When I put the sdcard into the camera it seems to ignore it; the firware was always the same 3.5.8_0406 :(

telmomarques, 2020-04-08:
Hi,

Try another SD Card, that usually solves the issue.

memma89, 2020-04-08:
Wow, you're right! I managed to hack my camera, thank you so much!
Can you tell me how can I access to the rtsp stream?

telmomarques, 2020-04-08:
RTSP is still being tested in closed alpha, will be released in the short-term.

memma89, 2020-04-08:
So wonderful! If there is a way to donate to support your hard work please tell me

telmomarques, 2020-04-08:
If you have any technical background and want to help, you can join us on telegram!
https://t.me/mijav4RTSP