telmomarques/xiaomi-360-1080p-hacks #26: I have managed to un-brick my MJSXJ05CM  with https://drive.google.com/file/d/1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj/view

Shevbo, 2020-04-17:
I have managed to un-brick my MJSXJ05CM  with https://drive.google.com/file/d/1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj/view

Any update on the hack?

_Originally posted by @Fruityski in https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/18#issuecomment-598550451_

absolutely the same - waiting for update for MJSXJ05CM to use RTSP!!!!

jonathanhar, 2020-04-18:
Same here, waiting for the RTSP :)

PauloDIas12, 2020-05-20:
Sorry. New in all this. I have a Mi Home Security Camera 360 1080p about two days.
I need to use it to mnonitor my baby. The Mi home app close in background and I need to keep it alive to hear the sound. What should I do?
Regards,

kathgironpe, 2020-05-24:
> Sorry. New in all this. I have a Mi Home Security Camera 360 1080p about two days.
> I need to use it to mnonitor my baby. The Mi home app close in background and I need to keep it alive to hear the sound. What should I do?
> Regards,

You better read a lot before trying things, because some have bricked their cameras. This repo is only for this version: `MJSXJ02CM`.

PauloDIas12, 2020-05-25:
> 
> 
> > Sorry. New in all this. I have a Mi Home Security Camera 360 1080p about two days.
> > I need to use it to mnonitor my baby. The Mi home app close in background and I need to keep it alive to hear the sound. What should I do?
> > Regards,
> 
> You better read a lot before trying things, because some have bricked their cameras. This repo is only for this version: `MJSXJ02CM`. I have read some who tried and got things to work for `MJSXJ05CM`.

Thanks

closed 2020-06-09 by telmomarques

dm800pvr, 2021-01-19:
Hey,
3.5.1_0052 downgrade is not possible for MJSXJ05CM !!!!
 I test several sd cards it just rename tf_recovery.img to tf_recovery.img.bak and create some folders!!
that's it! any solution?

mathieuedet, 2021-02-25:
Hi, same problem, I bricked my cam by testing downgrade and tested with two micro sdcard (2 sandisk classe 10) : a 16Go / a 64Go. I have formatted sdcard to FAT32 format. I have tested about 10 img but still solid yellow led... Any solution ?