telmomarques/xiaomi-360-1080p-hacks #27: List of which Models/ARM processors hack is needed?

marcusademola, 2020-04-23:
Perhaps below list is helpful and would be great if you can help to populate. Especially the ARM processor list (if you opened the device please include also photo). 

Model   (label) | Product name | FCC ID | FCC Date | ARM processor
-- | -- | -- | -- | --
CMSXJ19C | IMI Home Security   Camera 1080p Global | 2APA9-IPC019C | 10/23/2019 |  
CMSXJ16A | IMILAB Home Security   Camera Basic | 2APA9-IPC016A | 11/19/2018 |  
IPC019D | Smartthings Cam 360 | 2APA9-IPC019D | 03/09/2020 | Star SSC323 APC983UA   193J ARM
HMI205A | Mi Smart Plug Wifi | 2APA9-HMI205A | 04/27/2018 |  
CMSXJ25A | IMILAB EC3 Outdoor   Security Camera | 2APA9-IPC025B | 04/15/2020 |  
MJSXJ02CM** | Mi Home Security   Camera 360° 1080P | 2APA9-IPC009A | 08/21/2018 | Star SSC323 ACP025UA   193S ARM
CMSXJ19E | IMILAB Home Security   Camera A1 | 2APA9-IPC019E | 03/10/2020 |  
CMSXJ11A | IMILAB Wire-Free Home   Security Camera | 2APA9-IPC011A | 07/31/2019 |  
CMSXJ11AG | IMILAB Wire-Free   Camera Gateway | 2APA9-IPC011AG | 07/05/2019 |  
  |   |   |   |  

(status 23.4.2020)

** = Originally the hack was done for just one model, maybe its possible to expand a little. 
Maybe this table can be added on wiki page, including status:)

angadsingh, 2020-11-13:
do you know if the hack works with SXJ02ZM?

marcusademola, 2020-11-13:
> do you know if the hack works with SXJ02ZM?

I dont sorry, anyone else?
If not would you like to give a try, document it well and publish here

Povezi, 2020-11-16:
Does it work with CMSXJ25A ?

teras, 2020-12-14:
This guy opened the CMSXJ16A device...
https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/12#issuecomment-734817542

cstrassburg, 2021-06-07:
CMSXJ19E uses a Star SSC 323 ARM too: 

https://github.com/cstrassburg/cmsxj19e-hacks/blob/main/images/IMG_20210607_164115.jpg

gelsoniezzi, 2021-06-27:
CMSXJ25A - SSC325 AY00255B 2010J AMR
[image]

sihaizhishu, 2021-11-18:
> Does it work with CMSXJ25A ?

No. I have tried this on IMILAB CMSXJ25A. It is softbreaked it :(

marcusademola, 2021-11-18:
Thanks for the inputs, updated the table, but its still not clear which models were succefully flashed (entered)  so I didnt add hack-status column (yet). Feel free to input.

Meanwhile, I also discovered block diagram that might be useful. https://linux-chenxing.org/infinity6/ssc326d_blockdiagram.png

pixjuan, 2021-12-19:
Here is what I manage to decipher for my MJSXJ10CM:
SSC323
AP10DE8B
2114F-IM12
ARM
[image]

pixjuan, 2022-06-28:
> 
> can be hacked too? With rtsp?

The RTSP hack doesn't work out of the box. I have time this summer, I will try to get it working

0x7c9, 2022-06-28:
> > can be hacked too? With rtsp?
> 
> The RTSP hack doesn't work out of the box. I have time this summer, I will try to get it working

okay , i will wait for it.

Cavairs, 2022-07-09:
> > тоже можно взломать? С ртсп?
> 
> Взлом RTSP не работает из коробки. У меня есть время этим летом, я постараюсь заставить его работать

waiting for. would be great

thanasis00, 2022-07-15:
+1 from me as well! Waiting for updates, I can also help with testing and/or development.

emodtibor, 2022-09-14:
> > can be hacked too? With rtsp?
> 
> The RTSP hack doesn't work out of the box. I have time this summer, I will try to get it working

Twicer, 2023-12-05:
IMILAB C30 (looks like C21)
[image]
[image]
STAR SSC337DE AS0081B 2048 ARM

Detects as : chuangmi.camera.021e01 HW: 021501_1.6.0_0077
Open Ports:
53 domain
67 dhcps
684 cobra-iiop-ssl
772cycleserv2
959
8001 vcom-tunnel
9103 bacula-sd
17494
17638
18228
19719
20425
26720
32777 rpc18
39217
42172
45818
49180
49393
51255
53571
57843

NAME=Buildroot
VERSION=2017.08-00349-gb8cf9fb-dirty
ID=buildroot
VERSION_ID=2017.08
PRETTY_NAME="Buildroot 2017.08"
MODEL_COMMIT=87f5fc5
COMMON_COMMIT=b8cf9fb
CHUANGMI_VERSION=021501_1.6.0_0077

[+ +++ ---]  ali_stream          uptime: 406s/-s  pids: 712/-
[+ +++ +++]  fetch_av            uptime: 406s/406s  pids: 715/714
[+ +++ ---]  imi_agent           uptime: 406s/-s  pids: 719/-
[- --- ---]  imi_ota             (service not activated)
[+ +++ +++]  miio_client_helper  uptime: 406s/406s  pids: 723/722
[+ +++ ---]  miio_devicekit      uptime: 406s/-s  pids: 725/-
[+ +++ ---]  miio_record         uptime: 337s/-s  pids: 1074/-
[+ +++ ---]  miio_sdcard         uptime: 406s/-s  pids: 727/-
[+ +++ ---]  mortox              uptime: 407s/-s  pids: 703/-
[+ +++ ---]  mstar_algo          uptime: 337s/-s  pids: 1078/-
+ cat /run/nas/debug
cat: can't open '/run/nas/debug': No such file or directory
+ ls -hAl /mnt/sdcard/MIJIA_RECORD_VIDEO
ls: /mnt/sdcard/MIJIA_RECORD_VIDEO: No such file or directory
+ log_diag_platform.sh
/mnt/data/bin/log_diag.sh: line 1: log_diag_platform.sh: not found