telmomarques/xiaomi-360-1080p-hacks #33: Problem downgrade firmware

saipafo, 2020-05-28:
Hello, i followed your instructions but i was unable to downgrade the firmware.
When I put the sdcard into the camera, the led is solid yellow. And nothing happen. I'm waiting for 1hour...

I try to power off, and remove the SDCard, but same, nothing happen. Now i can't power on the camera.

I have try an other SD card. Same issue :( .

Can help?

telmomarques, 2020-05-28:
Hi,

Please confirm camera model on the bottom of the camera.

saipafo, 2020-05-28:
Yes!

Model: MJSXJ05CM

telmomarques, 2020-05-28:
Hey,

Check the supported camera model list on the repo readme 😋

See #18 how to unbrick your camera, join the discord server if you want more info: https://discord.gg/qggupzu

closed 2020-05-28 by telmomarques