telmomarques/xiaomi-360-1080p-hacks #35: 10min flashing ?

ysoudy, 2020-06-09:
Hi,

I insert the SD CARD with he downgrade firmeware, and I m waiting since 10min.
It seems very long for a flash :(

What should I do ?

ysoudy, 2020-06-09:
It's still on solid LED yellow

kakemanx, 2020-06-09:
Are your camera the right one?
I did the same thing... flashed a mjsxj05cm and bricked it (can get it back thou :) )
The hacked firmware is for the

ysoudy, 2020-06-09:
oops... my camera is Xiaomi Mi Home Security Camera 360
it's not the same ?

ysoudy, 2020-06-09:
How can I unbrick it ?

kakemanx, 2020-06-09:
> How can I unbrick it ?

Check the bottom of the camera and check that You have the 05 model, then download what i linked and flash it just like you did the other :)

ysoudy, 2020-06-09:
It's a 05cm :(
SO i just turn OFF, put the new firmware you give me from google drive to the SD card, and turn on ?

Thx a lot !

closed 2020-06-09 by telmomarques