telmomarques/xiaomi-360-1080p-hacks #39: Hack is not works

vitt76, 2020-06-15:
Thank u for great job! 
But for my camera hack isn't works.
I successfully downgraded firmware, but then I copy the folders from sdcard folder and put the card into the camera, after camera starts, is nothing happens. In MiHome I see the camera and picture and nothing in my IP address. In this case, MiHome said that my sdcard is corrupted. I tryed format sdcard several times, the it is empty, MiHome see the card, then folders present, MiHome said that is corrupt. What can I do for this?

vrajpatel2312, 2020-06-15:
you should try formatting SD using MI 360 cam, then insert sd card to pc to copy old firmware

vitt76, 2020-06-16:
I have no problem with old firmware, I wrote about the hack - "hacks" and "manu_test" folders