telmomarques/xiaomi-360-1080p-hacks #41: How can I add the camera to my lovecale?

winnertako, 2020-06-15:
How can I add the camera to my lovecale?

I've tried this way:

`type: picture-elements
image: /local/testpattern.jpg
elements:
  - type: image
    entity: camera.backyard_camera
    camera_image: camera.backyard_camera
    camera_view: live
    style:
      left: 50%
      top: 50%
      width: 100%
      height: 100%`

But I can't get any image...

HepoH3, 2020-09-24:
What camera do you use? If you have rtsp stream with h265 codec, it will probably don't work, see this [comment](https://community.home-assistant.io/t/add-support-for-h-265-in-stream-component/205661/16?u=hepoh).
Well currently I stack with the same problem.

winnertako, 2020-09-25:
Hello @HepoH3. I'm trying to use the xiaomi MJSXJ05CM. Thank you

HepoH3, 2020-09-25:
> 
> 
> Hello @HepoH3. I'm trying to use the xiaomi MJSXJ05CM. Thank you

So that's the case. We can only wait for the h264 support, cause it's very unlikely that h265 will be supported in browsers.

winnertako, 2020-09-25:
I've tried again today and it's working fine with this setup:

[image]

`- platform: generic
  still_image_url: "/local/testpattern.jpg"
  stream_source: "rtsp://IP_CAMERA:8554/mainstream"
  name: "Stream #1"`

closed 2020-09-25 by winnertako

thewh1teagle, 2020-09-25:
@winnertako 
Can you share configurations files? I want to add it into readme section

HepoH3, 2020-09-25:
> 
> 
> I've tried again today and it's working fine with this setup:
> 
> [image]
> 
> `- platform: generic still_image_url: "/local/testpattern.jpg" stream_source: "rtsp://IP_CAMERA:8554/mainstream" name: "Stream #1"`

Do you just grab files from the archive that been published in discord server and placed them into sdcard, or you following steps from [this discussion](https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/18)?

danergo, 2020-10-14:
This would interest me too. you might not have MJSXJ05CM but MJSXJ02CM. Please confirm.