telmomarques/xiaomi-360-1080p-hacks #55: Stuck while downgrading firmware

SweAdelpha, 2020-08-12:
Hi,
Thank you for your awesome work on this hack!
I'm having some trouble downgrading the camera though, it has been stuck with solid orange(yellow?) led for over 40 minutes now. Hopefully I'm just impatient since the instructions says "several minutes" which I guess could be under an hour.
Also I'm unsure about the "power down" procedure, this just means pull the power cord? I could only find a way to put the camera to sleep in the mi home app.

Would it be safe to pull the power cord and try again with a different SD-card?

telmomarques, 2020-08-12:
Hi,

Please confirm the camera model on the bottom of the camera.

SweAdelpha, 2020-08-12:
Sorry, I am a moron.
I've got the MJSXJ05CM.

Got it working again with the recovery file from this link:

https://c.mi.com/thread-3131455-1-0.html

closed 2020-08-12 by SweAdelpha