telmomarques/xiaomi-360-1080p-hacks #58: not accessible with http

xzonz, 2020-08-31:
Hello , thanks for the hack , i was waiting for it since months !!

i've downgraded the firmware -> ok  (verified on the android app mi home)

i've loaded the hack following the instructions , but , impossible to connect with http
browser said : '' ERR_CONNECTION_REFUSED ''
i've tried ssh with putty : same result.

i've tried with two differents sdcard !!
any suggestions doctor ? :-)

Thanks for your help

telmomarques, 2020-09-01:
Hi,

Please confirm you downloaded the latest release and not the source code.

xzonz, 2020-09-01:
hi , i took the **v1.2-rc1** ... right ?

telmomarques, 2020-09-01:
Yes, but make sure you downloaded "sdcard.zip" and not the source code.

xzonz, 2020-09-01:
ok , sure that's what i've downloaded.
i put the whole folders (framegrabber , rstp-server ..... )
at the root of the sdcard and followed the instruction.

telmomarques, 2020-09-01:
The folders "hacks" and "manu_test" should go no SDCard root.

xzonz, 2020-09-01:
i've just put those ones at the root

[image]

telmomarques, 2020-09-01:
That's the problem 🙂
You should unzip sdcard.zip directly on the SDCard.

xzonz, 2020-09-01:
ok i will make the test tonight

thanks very much for your help !  👍

pedromalima, 2020-09-19:
Hi,

I have a similar issue, I can view the feed from MI app, but when scanning through NMAP it does not work. all ports closed.
I got the following debug info
[miio_client.log](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/5250469/miio_client.log)
[diagnosis.txt](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/5250470/diagnosis.txt)
[messages.txt](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/5250472/messages.txt)

Can you please have a look?
the downgrade worked fine, thanks

titusfx, 2020-10-03:
> ok i will make the test tonight
> 
> thanks very much for your help !

Any luck?

Spirineu, 2020-10-14:
Hi, 

Thank you for this hack.

But I'm really new to HA, can you help me out?

Ive put the files on the sd card but I cant acess the web config using my I.P 😞 

[image]

Rtamagnini, 2020-11-18:
my camera works at ha with rtsp.
how to wotk via http ? 

http://192.168.1.104:8554/mainstream ?

Vorlex, 2020-11-18:
@Rtamagnini 
[image]