telmomarques/xiaomi-360-1080p-hacks #63: Compatibility with Mijia 1080p MJSXJ01CM?

davidrebe, 2020-09-29:
Hi!
is there any chance that this hack is compatible with model MJSXJ01CM?

[image]
[image]

Thanks!

bradyramone, 2020-11-09:
Not for me. It actually bricked it, was only able to resolve it following this post:
https://c.mi.com/thread-2609368-2-1.html

foofighter69, 2020-11-18:
> Not for me. It actually bricked it, was only able to resolve it following this post:
> https://c.mi.com/thread-2609368-2-1.html

Can you share the firmware? I can't download it.

bradyramone, 2020-11-18:
> > Not for me. It actually bricked it, was only able to resolve it following this post:
> > https://c.mi.com/thread-2609368-2-1.html
> 
> Can you share the firmware? I can't download it.

It's this one:
https://drive.google.com/open?id=1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj

davidrebe, 2020-11-19:
What do I have to do? How do I install that firmware?

SuperSpe, 2021-01-18:
I managed to get that working with a different firmware, but now I cannot find it anymore.
the rstp was poor too btw....
the version btw was 3.3.6_2017092810

SuperSpe, 2021-01-19:
found it 
https://www.dropbox.com/s/7lupgkyr3x0hbpu/Mijia%20720P%20RTSP%20hack%200.95%20and%20firmware.zip?dl=0

chand7704, 2024-01-14:
Hello

fufurax, 2024-04-03:
> MJSXJ01CM

Hi, it works on MJSXJ01CM ?

Tks