telmomarques/xiaomi-360-1080p-hacks #65: Not being able to downgrade

titusfx, 2020-10-04:
I've tried with 3 different SD cards, with different sizes 8Gb, 32Gb and 64Gb.
I formatted the SD Cards (just in case) from Mi Home, on the Camera options> Manage storage > SD card status > Format SD card.
Turn off the camera and extract the SD_Card
Copy the `tf_recovery.img` file to the root of the sd card
Insert the SD card to the camera and turn on the camera
It does the yellow thing, It keep Yellow for a minute, and it does the calibration (the 360 rotation on the X  and 120 rotation on Z)
Then, I still have access from Mi Home and when I go to firmware updates options (on my Mi Home) is the same firmware (which currently is `4.0.9_0409`)

In any case I tried to do the hack (copying the folders on the root of the SD card `hacks` and `manu_test` and deleting the `tf_recovery.img`, just in case) and nothing.

Could someone send me a picture of the type of sd card that are you using? In case the assumtion of some sd cards works and some other doesn't is true.

Or

Could someone tell me if they would be able to downgrade the camera with the latest version which is `4.0.9_0409` ? So I can be sure that the only problem is the sd card.

Thanks

Phoenigore, 2020-10-17:
Hi, I've got the same issue. Tried with 2 sdcards

rezmus, 2020-10-17:
you can't downgrade from 4.0.9_0409. they replaced uboot and rsa keys with this update so none of old recovery firmwares can be used. it's closing holes after imilab source code leak.

sergioska, 2020-11-15:
I've the same problem. Maybe there is a workaround?

GdLoQ971, 2020-11-17:
the same as you too

Rtamagnini, 2020-11-17:
works for me with .
 my model :
[image]

[image]

sd 8gb

my firmware version ;

[image]

Vorlex, 2020-11-17:
I was able to downgrade too using my old Nokia 64 MB MicroSD. I wasn't able to downgrade using one that is used in camera (32GB). I've read on one [Russian site](https://sprut.ai/client/blog/921) that it won't update using cards with capacity more that 16GB. And you better use no-SDHC cards. Author on that Russian site said he used 16GB SDHC, though.

sergioska, 2020-11-17:
> my firmware version ;
> 
> [image]

@Rtamagnini where can i find that firmware version? can you post a link please?

Vorlex, 2020-11-17:
@sergioska it is in readme :)

sergioska, 2020-11-17:
> @sergioska it is in readme :) [link](https://github.com/telmomarques/xiaomi-360-1080p-hacks/raw/master/firmware/3.4.2_0062/tf_recovery.img)

sorry, i thought that it was another firmware version. Repo firmware doesn't work in my case. I tried with 2 sd card 2G and 8G, but nothing.

Rtamagnini, 2020-11-17:
4.0.9_0409 version is for MJSXJ02CM ? 
I suppose not. because the last one that is shown to me is 3.4.6 ...

Vorlex, 2020-11-17:
@Rtamagnini, it's because of incremental update. Camera can't be updated straight to the latest version. It updates to 3.4.6_0213, then to next one (maybe 4.0.9_0409).

Vorlex, 2020-11-17:
@sergioska, did you try to format to FAT/FAT32?
Here's photo of my card that worked (FAT formatted):
[image]

NeWoKiKi, 2020-12-09:
Hello,

I have MJSXJ02CM in 4.0.9_0409.
So, is it possible to downgrade or no ? @Rtamagnini / @Vorlex were you in 4.0.9_0409 before downgrading ?

I tried with 2 different sdcard but it's not working for me.
The camera is booting and (probably) ignoring downgrade.

I would like to know if it's possible before trying all sdcards in the world :D

Many thanks,

rezmus, 2020-12-09:
not possible, because uboot was replaced with this firmware. it use tf_update.img signed with different keys.

Vorlex, 2020-12-09:
@NeWoKiKi
I'm sure I was on the last firmware before applying hack. Do you know when 4.0.9_0409 was released?

rezmus, 2020-12-09:
it depends on region. 4.0.9_0409 was only in ota on cn region. others have older firmware.

NeWoKiKi, 2020-12-09:
I'm in France and using EU servers on Xiaomi Home, I don't now when but my camera self updated to 4.0.9_0409

@titusfx created this post on october 4th so... firmware was released at least 2 months ago :D

rezmus, 2020-12-09:
i just checked and some accounts on eu region show ota to 4.0.9_0409, while some still shows 3.4.6_0213 as last version. however it should not update by itself. sometimes they display popup when you enter camera so it's easy to click it and update.

anyway no way to downgrade from 4.0.9_0409 unless you open device and use programmer.

Vorlex, 2020-12-09:
I'm on the Chinese servers and pretty sure I had last firmware. There is no way to check now, but I successfully applied the hack.

GdLoQ971, 2020-12-09:
> anyway no way to downgrade from 4.0.9_0409 unless you open device and use programmer.

I looked a little on this side, without much result :( @rezmus do you have some info ?

rezmus, 2020-12-09:
if your uboot accepted tf_recovery.img you could not have 4.0.9_0409, because new uboot is only checks for tf_update.img

VinceNM, 2020-12-13:
> if your uboot accepted tf_recovery.img you could not have 4.0.9_0409, because new uboot only checks for tf_update.img
> 
> https://ufile.io/v41nwo40
> 
> @GdLoQ971 i never had to open, but you can find some information in MJSXJ05CM issues

Do you have some informations on how to downgrade when we are on 409 ?

GdLoQ971, 2020-12-13:
@VinceNM Sorry, but so far nothing I have tried has worked

carolinechap, 2020-12-13:
> @VinceNM Sorry, but so far nothing I have tried has worked

What did you actually do to make it work ?

VinceNM, 2020-12-13:
Je suis dépité d’avoir Mis à jour mes 3 caméras 😭

Vincent

GdLoQ971, 2020-12-13:
Pareil mais pour 4 ;) courage, nous trouverons

carolinechap, 2020-12-13:
J'en ai mis qu'une à jour et heureusement.. :D Si vous avez des pistes! N'hésitez pas à partager!

GdLoQ971, 2020-12-13:
> J'en ai mis qu'une à jour et heureusement.. :D Si vous avez des pistes! N'hésitez pas à partager!

oui, mais pour l'instant très peu de piste, malheuresement...

abdullahgunduz, 2020-12-15:
Why do you want to downgrade 4.0.9_0409. What is 4.0.9_0409 problem?
And what is changed in 4.0.9_0409 any body know?

carolinechap, 2020-12-15:
@abdullahgunduz You can't hack the camera if you already have the 4.0.9_0409 firmware version.

abdullahgunduz, 2020-12-15:
Its a bug or fix i cant understand.
What do you mean "hack".

carolinechap, 2020-12-15:
The purpose of this repo.. Using the IP of the camera and avoiding using the xiaomi app (to be simple)

toannguyen1412, 2020-12-19:
> not possible, because uboot was replaced with this firmware. it use tf_update.img signed with different keys.

I think about that, i try 3 SD-Card but not working.

UmutAlihan, 2021-01-03:
> update.

so do you think if I change tf_recovery to tf_update, will it start downgrading??

rezmus, 2021-01-03:
you can't downgrade using sd card recovery from 4.0.9_0409 (MJSXJ02CM) or 4.0.9_0426 (MJSXJ05CM). with these firmware versions they replaced uboot/keys. older recovery firmwares won't work. you can only downgrade with programmer.

titusfx, 2021-01-03:
@Vorlex if you don't mind trying and loose the hack, I mean, if you are sure you did the downgrade with 4.0.9_0409, you could update to 4.0.9_0409 and then apply the hack, and let's know that you did it. I video will be really appreciated

titusfx, 2021-01-03:
@rezmus do you have a link, how we can use the programmer to hack the camera? Thanks

buttermocca, 2021-01-04:
https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/79

MJSXJ02CM firmware chip can write on CH341A rom writer???

sandeep-shyam, 2021-01-06:
can anyone share the tf_recovery.img file for 4.0.9_0409, my camera is stuck on yellow LED, i guess its due to power cut during firware download or upgrade

UmutAlihan, 2021-01-06:
mine fails to downgrade :/

it shows yellow light for a while than tells me waiting for qr code. After
connection it still shows firmware 4..

sonnt136, 2021-01-07:
> can anyone share the tf_recovery.img file for 4.0.9_0409, my camera is stuck on yellow LED, i guess its due to power cut during firware download or upgrade

you can try https://drive.google.com/file/d/15i3xL7Q-ZjR9nOgVtqSogx1HjDbsFAZB/view?usp=sharing

sandeep-shyam, 2021-01-07:
Great , thank you so much

buttermocca, 2021-01-07:
> > can anyone share the tf_recovery.img file for 4.0.9_0409, my camera is stuck on yellow LED, i guess its due to power cut during firware download or upgrade
> 
> you can try https://drive.google.com/file/d/15i3xL7Q-ZjR9nOgVtqSogx1HjDbsFAZB/view?usp=sharing

really great thank you. i'll try to install it.

buttermocca, 2021-01-07:
https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/79

I've already tried to recover with this version of firmware.
Someone else has answered that the unique identification number has been erased from my firmware chipset.
I guess there was a problem because I tried to recover it with the firmware of another product MJSXJ05CM.

sonnt136, 2021-01-08:
> #79
> 
> I've already tried to recover with this version of firmware.
> Someone else has answered that the unique identification number has been erased from my firmware chipset.
> I guess there was a problem because I tried to recover it with the firmware of another product MJSXJ05CM.

You can try this file https://drive.google.com/file/d/1L18wbWuk-41O0NqG0YQElmxUkZ5YPwQ7/view?usp=sharing

UmutAlihan, 2021-01-09:
> I'm on the Chinese servers and pretty sure I had last firmware. There is no way to check now, but I successfully applied the hack.

Can you apply the hack and live happily ever after without downgrading the firmware, with updated FW 4+??

mfigueroa7, 2021-01-30:
The same problem. I have the 4.0.9_0409 version in China Maindland and i can´t downgrade the version. Always the camera is in the4.0.9_0409 version. Any workaround for this camera version?

133usr, 2021-02-01:
Here too . .. 
I have the 4.0.9_0409 version in China Maindland and i can´t downgrade the version. any way ? to downgrade the version??
Plz help

VinceNM, 2021-02-13:
Any workaround for this issue ? I'm still stuck on the 4.0.9_0409...

ptzenos, 2021-02-27:
Same issue for me, I can't downgrade. Is there anything we can do, even if it means that we have to disassemble the camera?

shiekhathar, 2021-03-06:
It is such a pain, since the new update **I have stopped receiving notification on my mobile device for the camera. Its specific to iOS because android devices get the notifications for motion.**
Also, the firmware version is 4.0.9_0426. I have 3 cams and none give me a push notification. _I thought of downgrading but as I can see that it is not possible to downgrade from this version_. _Any idea how to fix this_? 😭😭

romalb47, 2021-04-28:
I dump the 4.0.9 flash and extract it. And indeed the Factory file have change a lot (with rsa signature check, etc...). I change the script to just run an file on the sdcard and repack the squashfs and it seem to do the job.
Next thing: try the hack to see if it work.

Some one know if there are sensible data onto the flash (at reset state)? Like an unique id.

rinorbytyci, 2021-05-01:
> I dump the 4.0.9 flash and extract it. And indeed the Factory file have change a lot (with rsa signature check, etc...). I change the script to just run an file on the sdcard and repack the squashfs [...]

Does it work with the latest version? Do you have files to share? I am on 0426 is there any hope to downgrade or install the hacks?

informadortech, 2021-05-12:
> I dump the 4.0.9 flash and extract it. And indeed the Factory file have change a lot (with rsa signature check, etc...). I change the script to just run an file on the sdcard and repack the squashfs [...]

ANy update on this? can you share the files?

truonggiang152, 2021-06-02:
any update with fw 4.0.9_0409 @@?

Zeunas, 2021-06-08:
> I dump the 4.0.9 flash and extract it. And indeed the Factory file have change a lot (with rsa signature check, etc...). I change the script to just run an file on the sdcard and repack the squashfs [...]

Got anywhere with this? Would be great if we could hack the most recent versions.

belkone, 2021-06-08:
it is no longer possible to downgrade with an SD card.
hacking newer firmware versions is possible with the spi flasher: https://gist.github.com/belkone/fb63376a9b6b6ebe969e9c8f96c0defb

@Zeunas you shouldn't share your flash dump as it contains sensitive data, but it is possible for everyone to modify their own dump ;)

hacks works (no h264 streaming in rtsp, h265 only).

Souroshis, 2021-06-22:
I am From India 
I have MJSXJ02CM From 1 Year I am Not Able to use the camera 
Last time I think the update was going on and there was an electric cut after that the camera turns on with steady yellow light and makes calibration (the 360 rotation on the X and 120 rotation on Z) after that it doesn't pair with the mi home . It doesn't reads the Qr code at all. 
I have tried Reset it with the button Then it says something in Chinese and makes  calibration (the 360 rotation on the X and 120 rotation on Z) 
Then again went to steady yellow light. 
Please Anyone help Me :( @titusfx @rinorbytyci @romalb47 @sergioska @VinceNM

vnlove, 2021-08-17:
Any workaround for this issue ? :((

zhangguiyu, 2021-12-25:
I am on 4.0.9_0409 and it is stuck on yellow light.

Flashed with 4.0.9_0409 version of tf_recovery.img does not fix it.

@rinorbytyci : wondering if you are able to share your tf_recovery.img.backup (4.0.9_0426) from your sdcard (if you uses one).

Thank you so much!
Julian

> @zer0-infinity

Wondering if you have a copy of yoru tf_recovery.img (or backup version) corresponding to 4.0.9_0426?

[4 more replies]