telmomarques/xiaomi-360-1080p-hacks #68: Camera stayed with the yellow led forever

guiguille, 2020-10-23:
When trying to install the firmware with the SD card, camera led stayed yellow and nothing else happened.
After 25min, I shutdown it a tried without the SD card. Still same, so now my camera is useless can't use anymore ?

Anyone has a solution ?

Karwail, 2020-10-23:
Quel est la version de ta caméra ? (Sous la caméra, pas sur la boîte)

guiguille, 2020-10-23:
MJSXJ05CM

Karwail, 2020-10-23:
Tu as brick ta caméra ... J'ai eu même problème (numéro sous la caméra n'est pas la même boîte..)
Il faut maintenant fouiller sur le web pour trouver le firmware de la mjsxj05cm

Karwail, 2020-10-23:
https://karwail.freeboxos.fr:5001/share/tLAykxjyVWNQvWVo/

Essaye le firmware nommé IPC .... donne moi des nouvelles !!

guiguille, 2020-10-23:
Merci de ton aide, j'avais pas vu ton 2 eme commentaire! J'ai finalement après mon message, trouvé sur un forum russe obscure un firmware assez ancien mais qui fonctionne! Tout est rentré dans l'ordre 😄

closed 2020-10-23 by guiguille