telmomarques/xiaomi-360-1080p-hacks #69: Exchange the voice set in firmware

vkcorp, 2020-11-11:
I'm using a model of MJSXJ08CM and could get a firmeare from the help of rezmus (thanks),
and trying to change the china set to english.

There are 2 set of voice packages like :
    - sound/en
    - sound/cn

And there are some symlic link on top of sound/ folder like :

    lrwxrwxrwx 1 root root 15  8월  1 15:57 binding_fail -> cn/binding_fail

It's linked to sound/cn files for now,
but if I will change the symbolic link to sound/en maybe I can hear the voice via english.
Not sure yet but I hope so ;)

After getting my confirmation, then I will try to chnage the voice set to another one like Korean.

From the extracted sound file, I think the sound file formart is:

    - (Raw) / A-Law mono / 16000

So I'm thinking I can make new one with that format !

If there are any help or information your have for me, please let me share it :)

vkcorp, 2020-11-11:
Actually I guessed there was a property to select the voice set (cn or en) in somewhere of script files,
but nothing there... May I didn't find yet or nothing from origin.
So I'm trying to change the voice file itself !

vkcorp, 2020-11-16:
Eventually I guess it's not possible to update new firmware on my MJSXJ08CM with 4.0.9_0188.
I've fixed and made new one (with fixing data region only from old firmware),
but my camera wouldn't accept new one with flashing. Not sure but I guess it's blocked to flash new one on my 4.0.9...

ranov-tr, 2021-02-13:
Hi, can you please share a working firmware?