telmomarques/xiaomi-360-1080p-hacks #72: MJSXJ02CM at Hassio

Rtamagnini, 2020-11-14:
hello,  I was able to easily downgrade my camera's firmware and hack.
via browser I put:
-enable RTSP and Websocket Stream
-Encoding type: H264
-Enable SSH / SFTP
-created a new username / password

now I see the camera with the address: "rtsp: //192.168.1.104: 8554 / mainstream" and "rtsp://192.168.1.104:8554/substream"

can you tell me what I have to do now to be able to see the camera on Hassio (rpi3)?

do I need to open any ports on my router? 
what instruction to put in yaml?

something like this ?
camera:
  - name: room
    platform: ffmpeg
    stream_source: "rtsp://admin:123456@192.168.1.104:554/h264Preview_01_main"  

Thanks in advance
Ricardo

Rtamagnini, 2020-11-14:
I also noticed that if I put the encoding on the h264 I no longer have an image on the Mi home app

Vorlex, 2020-11-15:
@Rtamagnini 
Here's configuration:
```
camera:
  - platform: generic
    name: Xiaomi 360
    stream_source: rtsp: //192.168.1.104: 8554/mainstream
    still_image_url: http://127.0.0.1:8123/local/camera.jpg
```
There is [integration guide](https://www.home-assistant.io/integrations/generic/) for generic camera where you can read more about it. 
Also, this camera doesn't provide static image. You can follow [these steps](https://community.home-assistant.io/t/camera-generic-set-still-image-url-to-local-file/130351/6?u=vorlex) to add custom image as preview on lovelace dashboard.

Rtamagnini, 2020-11-17:
i don't have any video or image.

do i need to put user/pass at rtsp command ? 

my config.yaml 

  - name: Oficina
    platform: generic
    stream_source: rtsp://192.168.1.104:8554/mainstream
    still_image_url: http://127.0.0.1:8123/local/camera.jpg

I put a camera.jpg file in the www directory

and in the view I put a card like this:

type: picture-glance
title: Oficina
image:
  camera_image: camera.Oficina
entities:
  - switch.sonoff_1000889b10

any help or any trace i can do ?

Rtamagnini, 2020-11-17:
WORKS!!!!

type: picture-entity
entity: camera.oficina
camera_view: live

wallylee99, 2020-12-30:
> WORKS!!!!
> 
> type: picture-entity
> entity: camera.oficina
> camera_view: live

I can't get the image to show on my home assistant (pi3).

My configuration.yaml:

camera:
  - platform: generic
    name: mi
    stream_source: rtsp://192.168.1.16:8554/mainstream
    still_image_url: http://192.168.1.106/cam.jpg

And I add a Picture Entity Card to the dashboard with following info:

type: picture-entity
entity: camera.mi
camera_view: live
show_state: false

Any help is appreciated!

wallylee99, 2020-12-30:
Answering my own question, I manage to resolve it by using the following in my configration.yaml:

  - platform: ffmpeg
    name: mi
    input: rtsp://192.168.1.16:8554/mainstream

Then re-adding a picture-entity works!

ethaniel80, 2021-03-31:
Good morning. I have the firmware version 4.0.9_0409. 
Can I still downgrade and then integrate the camera into the Home assistant? 
Does this project still work? 
thanks

milander, 2021-05-20:
Same question as above

8999400, 2021-11-29:
> > РАБОТАЕТ!!!
> Я не могу получить изображение для показа на моем домашнем помощнике (pi3).
https://github.com/AlexxIT/WebRTC

StanAmsterdam, 2022-01-02:
> Good morning. I have the firmware version 4.0.9_0409. Can I still downgrade and then integrate the camera into the Home assistant? Does this project still work? thanks

Same. Tried downgrading on my current SD (64GB with video on it too) but didnt do anything.
Is the current FW protected/fixed and not downgradable or need to try on another empty SD as suggested?

PS. Any option to use camera as normal (no RTSP, but usage in the Mi app and video files on SD), but only enable FTP (and SSH)?