telmomarques/xiaomi-360-1080p-hacks #73: Led stay solid yellow

moi22441, 2020-11-18:
I have follow the instruction but the led stay solid yellow during 2 hours.
I don't want to unplug the camera because it's wrote "⚠️ Do not power down the camera while flashing!"

What can i do ?

tanks for help

moi22441, 2020-11-19:
1 Days after the led still yellow :(

vkcorp, 2020-12-02:
What kind of instructions did you do ?

md-ray, 2020-12-09:
I have the same issue, during downgrading the firmware. It's been more than 2 hours, so I believe something is stuck

telmomarques, 2020-12-09:
The only supported camera is MJSXJ02CM. Do not flash any other model, or you will soft brick it.

To recover the camera you need to flash the original firmware again. Check the pinned messages on discord (badge on readme) to get the correct firmware.

md-ray, 2020-12-09:
Opps, sorry my bad for my stupid mistake. Seems like my model is MJSXJ05CM.
Thanks @telmomarques for quickly pointing out the original firmware page, I'm restoring it now.

moi22441, 2020-12-09:
Sorry, i have a MJSXJ05CM to, i have restoring it with another firmware
Thanks for your help

closed 2020-12-09 by moi22441