telmomarques/xiaomi-360-1080p-hacks #74: Strange situation with mjsxj09cm

KhArtNJava, 2020-11-27:
Hello.
I've already successfully started RTSP on MJSXJ05CM ( https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/18#issuecomment-734339211 ) and now I am working on mjsxj09cm.
My script is 

```
#!/bin/bash
# extract firmware
mkdir -p files

dd if=backup.bin of=files/kernel.bin bs=1 count=2097152 skip=327680
dd if=backup.bin of=files/rootfs.bin bs=1 count=7733248 skip=2424832
dd if=backup.bin of=files/data.bin bs=1 count=6488064 skip=10158080
dd if=backup.bin of=files/vendor.bin bs=1 count=131072 skip=16646144

# patch jffs2 partition
sudo modprobe mtdblock
sudo modprobe mtdram total_size=6336
sudo dd if=files/data.bin of=/dev/mtdblock0 bs=1
mkdir mount
sudo mount -t jffs2 /dev/mtdblock0 mount

echo '#!/bin/sh' | sudo tee -a mount/bin/log_diag_platform.sh

echo 'echo qqq >> /mnt/sdcard/qqq.txt' | sudo tee -a mount/bin/log_diag_platform.sh
echo '/mnt/sdcard/override.sh' | sudo tee -a mount/bin/log_diag_platform.sh

sudo chmod 755 mount/bin/log_diag_platform.sh

sudo umount mount
rmdir mount
sudo dd if=/dev/mtdblock0 of=files/data.bin bs=1

# update flash backup
mkdir -p out
cp backup.bin -f out/flash.bin
dd if=files/kernel.bin of=out/flash.bin bs=1 count=2097152 seek=327680
dd if=files/rootfs.bin of=out/flash.bin bs=1 count=7733248 seek=2424832
dd if=files/data.bin of=out/flash.bin bs=1 count=6488064 seek=10158080
dd if=files/vendor.bin of=out/flash.bin bs=1 count=131072 seek=16646144

sudo flashrom -p ch341a_spi -w out/flash.bin
```
The board of mjsxj09cm is same to MJSXJ05CM board

[image]
[image]

Here is dump of it 
[backup.zip](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/5609090/backup.zip)

The camera has different authorisation in Mi Home process - after QR code it saying in Chinese randomly generated 4 digit password, that you should type in Mi Home App. I used Google Audio Translate to understand those 4 digits.

The problem is that they are possibly changed something in firmware, because it can't start *.sh file from sdcard. 

I especially added line in the code
`echo 'echo qqq >> /mnt/sdcard/qqq.txt' | sudo tee -a mount/bin/log_diag_platform.sh` 
To check that it can access sdcard. In result, /mnt/sdcard/qqq.txt created with qqq text, so it can access sdcard.

But /mnt/sdcard/override.sh doesn't execute, in log/diagnosis.txt there is an error
```
+ log_diag_platform.sh
/mnt/data/bin/log_diag_platform.sh: line 3: /mnt/sdcard/override.sh: not found
```

Log folder archive 
[log.zip](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/5609118/log.zip)

May be they are locked execution of *.sh scripts on sdcard? How to make it run sh scripts?

And yeah, I checked a lot time, override.sh and busybox are on my sdcard, sdcard are recognizable in Mi Home app in camera settings, camera writing videos on it successfully. The similar configuration works well on MJSXJ05CM.

xrays72, 2020-12-07:
Привет, Артур.
Получается завести RTSP на mjsxj09cm?

KhArtNJava, 2020-12-07:
> Привет, Артур.
> Получается завести RTSP на mjsxj09cm?

Пока что не занимался, времени не было...
Постараюсь на днях время выкроить, собрать для ipc029 frame grabber.

xrays72, 2020-12-08:
Спасибо, очень бы хотелось RTSP получить с этой камеры, а если бы прошивка с хаком была и завести эту камеру в home assistant, то вообще бы сказка была =)

wangshuo2020, 2021-03-15:
Hello bro, can rtsp work on mjsxj09cm now?

cstrassburg, 2021-06-10:
@KhArtNJava 
If you have already read out the flash chip, why don't you change the init.d scripts directly? you can add a hook or call a telnetd / ssh inside the rootfs.

SubZero77, 2021-07-14:
> > Привет, Артур.
> > Получается завести RTSP на mjsxj09cm?
> 
> Пока что не занимался, времени не было...
> Постараюсь на днях время выкроить, собрать для ipc029 frame grabber.

Дружище привет! Я готов заплатить за взлом этой камеры. Купил себе 2 таких и понял что это развод от Xiaomi. Хочу чистую rtsp и onvif без всяких китайских серверов :) (5 т.р. за взлом готов отдать)

nickthegreek, 2021-07-27:
I assume you're using Windows to create the override script. You need to do EOL conversion. If you use Notepad++ go to Edit->EOF conversion -> Unix(LF).

Alternatively run `dos2unix -u override.sh`

dpolyakov, 2021-09-13:
@KhArtNJava this link may be help https://github.com/SungurLabs/sungurlabs.github.io/blob/6043366d497943e0a246a6a420ba8fb2adfcef31/_posts/2021-07-14-Xiaomi-Smart-Camera---Recovering-Firmware-and-Backdooring.md

ImBluue, 2021-11-22:
Hello! Do you have any update on this issue ? Maybe consider doing a fork like for the mjsxj05cm.
BR

WooDeNDark, 2021-12-30:
> 

Same problem. There is very little left, but here it is =(

darkfuncat, 2022-01-14:
Hi all,
I recently bought a MJSXJ09CM and custom firmware seems to be the only way to use this cam (dont want chinese servers).
I understand that there is a way to hack whith a serial wired disasembled camera, but is it possible to hack/flash the camera only via sdcard ?
I successfully hacked an old MJSXJ02HL with another project (full root access with only flash sd), but this model is much better.
Many thanks !

chicaneau, 2022-02-16:
@usehelloworld how did you connect the camera to your wifi? there doesnt appear to be anywhere to configure SSID and pass?
for some reason mine wouldnt connect with mi home, hence trying to use a custom solution. I have shell access over UART and managed to run your command sequence above, however no wifi makes it kinda useless.

Cheers

mrpg99, 2022-05-26:
Would love to get onvif / rtsp support on the mjsxj09cm model, is it possible with some form of sdcard hack?

moreni57, 2022-10-16:
one have the solution for hack RTSP MJSXJ09CM

Agha-Shadi, 2022-11-28:
@usehelloworld 
I use windows, I wonder if i can flash an sd card with your file and do the rest on a linux desktop then take off the sd card and connect the camera to windows, and still be able to use it as a webcam! is that possible dude!?

r-science, 2022-12-07:
> @usehelloworld I formatted the sd-card as a FAT partition and then copied your 6 files inside the root of it. unplugged the MJSXJ09CM camera and inserted the sd-card inside. plugged it again to the power and tried to `telnet local_ip 80` but it rejected the telnet. I use windows 10. telnet is enabled and usable on the machine. used two different sd-cards with no chance. any idea on how I can solve this issue?

remember to "cd /tmp/bin" before 4. step from [usehelloworld](https://github.com/usehelloworld) tutorial :)

glemag, 2022-12-22:
Can't get this camera to HA, any updates?

geldelo, 2023-01-02:
> Can't get this camera to HA, any updates?

Рабочего варианта по интеграции в HA нет. Единственное решение RTSP, но я не пробовал, предложил выше [usehelloworld](https://github.com/usehelloworld), но там паять надо…

geldelo, 2023-05-12:
https://github.com/hacefresko/Xiaomi-Mi-360-UART-shell

ARMarchetti, 2023-09-14:
Hi. Not sure if this project is complete and if you managed to get this working for the MJSXJ09CM. If so pls can you help me to get RTSP hack on my cameras. Sorry I am a NOOB to this so any help would be appreciated.

nik-prostok, 2023-10-28:
https://github.com/hatsujouki/xiaomi-camera-MJSXJ09CM-hack