telmomarques/xiaomi-360-1080p-hacks #75: Motor control

cadavre, 2020-12-14:
I can see new hack for motor control. BTW there's no release for it available on Github...

But the question is – how to use it? :)

There's no info in README of this repo or dedicated repo.

uberkael, 2020-12-19:
I'm also interested, I tried with some ONVIF  control programs with no luck.

p2baron, 2022-03-31:
A response to an very old issue but I was wondering the same. 
To get this working with onvif follow the instructions from [this repo](https://github.com/thewh1teagle/xiaomi-1080-360-motor-control) and add those hacks to your sdcard. (motor_control and onvif_srvd folders)

mcarbonneaux, 2022-05-21:
> To get PTZ working with onvif follow the instructions from [this repo](https://github.com/thewh1teagle/xiaomi-1080-360-motor-control) and add those hacks to your sdcard. (motor_control and onvif_srvd folders)

why not indicate this in the readme ? or simply integrate it like in the [mjsxj05cm-hacks](https://github.com/cmiguelcabral/mjsxj05cm-hacks) ?

p2baron, 2022-10-04:
And to do this from SSH you need to write to the event file

`echo "tilt forward 10" >> /mnt/sdcard/hacks/motor_control/event`
`echo "pan reverse 10" >> /mnt/sdcard/hacks/motor_control/event`