telmomarques/xiaomi-360-1080p-hacks #79: MJSXJ02CM can write firmware on CH341A rom writer??

buttermocca, 2021-01-03:
It became a brick while updating the camera firmware.
[image]
When I disassembled the camera, I can see a firmware chipset.
cFeon
QH128A-104HIP
I took the chipset off and wrote the firmware version 4.0.9_0409 on the ch341a and tried to operate again, but only the yellow light kept on coming out and the motor didn't work.
Before updating to version 4.0.9, I have attached the firmware file that I had backed up to the ch341 device and the 4.0.9 firmware file that I used to update.

4.0.9 0403
https://tinyurl.com/y77rcgyu

backup firmware
https://tinyurl.com/3uxwybh

Can anyone check what the problem is?

rezmus, 2021-01-04:
link to backup is damaged.

buttermocca, 2021-01-04:
> link to backup is damaged.

https://tinyurl.com/y7xcttj4
i re uploded.

rezmus, 2021-01-04:
it looks like cloud credentials are gone (0xff0000). did you make backup before flashing anything with programmer or after?

buttermocca, 2021-01-04:
> it looks like cloud credentials are gone (0xff0000). did you make backup before flashing anything with programmer or after?

There is only firmware backed up after the brick T.T

cstrassburg, 2021-06-17:
Are you solved your problem? Maybe I can help you.

Souroshis, 2021-06-22:
@cstrassburg Please help I am Having Same Problem

cstrassburg, 2021-06-22:
Really the same problem? You wrote the spi chip with a programmer and nothing is working now? Do you have a dump from the flash? A dump before the update?

pexcn, 2021-07-27:
Hi @buttermocca , can you send provide MJSXJ02CM ch341a full backup firmware for me? My MJSXJ02CM was bricked...

jimmycr, 2021-10-05:
here you are...
[4.0.9 for MJSXJ02CM - tested](https://uloz.to/file/bqMw0AwKvWhg/xiaomi-camera-360-1080p-mjsxj02cm-4-0-9-0409-bin#!ZGV3ZQR2ZJVlAJRlMJExMGp4AmLmZJqbI3LlBIMjHHc6MwAvLj==)

jimmycr, 2021-10-09:
If this is question for me - "AS Programmer 2.0.0 something". When I needed to program the chip I had to use "Unsecure->Erase->Write->Verify" and then I had to set Secure bits again. Without Securing chip, camera "died" again on second restart after proper setup.

zhangguiyu, 2021-12-25:
The reason the backup (and all older) firmware does not work is due to change in firmware formart from version 4.09 onwards. You will only be able to update with firmware newer than 4.09. There is no way to downgrade the firmware, apparantly.

L337C0D3R, 2022-02-14:
someone provide software required for ch341a to work, mine is showing device not detected