telmomarques/xiaomi-360-1080p-hacks #81: SD card type + format ?

Toukite, 2021-03-14:
Hello,

I am trying to follow the forked procedure https://github.com/cmiguelcabral/mjsxj05cm-hacks but we can't open issues on this repo :)
In the procedure it is said that there can be an issue with SD card preventing the downgrade, and that we should try with another SD.
I think I am in this case since I have the yellow light on for an hour but camera never starts rotating.

Question is, to avoid random buy of another sd card (because I don't have any other ^^), could you tell us in which conditions you managed to succeed in your case, for example:

- which SD card type do you recommend (SD / SDHC / SDHX / ...) ?
- which filesystem do you recommend we format the SD card with (NTFS / EXT4 / FAT32 / ...) ?

Also a remark, the procedure for MJSXJ05CM talks about tf_recovery**.bin** file, but I actually downloaded a tf_recovery**.img** when following the link. Should I rename the extension or is it a documentation error ?

Thanks for the help and the good work !

thoaitam, 2021-03-21:
I tried all .img and rename to .bin, all of them only solid yellow too, :(

margusanvelt, 2021-05-21:
Regardiing which filesystem to use - put the SD card in the camera and format it using the app Storage Settings > Storage status > Format SD card.  

But i do believe that with the newest firmware the downgrade is not possible more in this issue: https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/65