telmomarques/xiaomi-360-1080p-hacks #84: Not being able to downgrade from 4.0.9_0426 firmware version

Niptoc, 2021-06-02:
Hello,

I have tried with several SD card and the downgrade from firmware 4.0.9_0426 is not working.
The firmware is renamed as tf_recovery.img.back but the downgrade is not done.  Can you help me ?

Regards,

rinorbytyci, 2021-06-08:
I think it doesn't work with the later firmwares anymore to downgrade sadly

belkone, 2021-06-08:
@Niptoc @rinorbytyci check it out: https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/65#issuecomment-857015177

chimeranzl, 2021-06-10:
I bricked mine updating from 3.x to 4.x, I can't even recover it with the v4 of tf_recovery.img though.  It always goes back to orange light, or sometimes orange flashing light.  I did get it working twice (unsure what I did) but each time when I try to pair up to mi home it can't find the wifi network.  Feel like throwing this thing in the trash.  Only just bought it too, haven't had a chance to use it yet.  Waste of money.