telmomarques/xiaomi-360-1080p-hacks #85: No rstp

panoskpv, 2021-07-13:
I  have  4.0.9 firmware and used this https://gist.github.com/belkone/fb63376a9b6b6ebe969e9c8f96c0defb. I used files in the latest release. The ones in "download zip" option not working at all. I can access the web UI but no rstp. In web UI I get a loading circle like YouTube, and when I go to rstp URL using VLC I get an error. Also, if I add onvif support I can move camera using ptz, but I have no image. Any ideas?

developergray, 2021-07-26:
Exact same issues with mine too. I used the programmer with same guide. Can access the webpage, but stuck at loading spinner. App streams ok. 

 Is it possible the newer firmware is blocking the access? If so, is it possible to downgrade to older versions?

L337C0D3R, 2022-02-19:
same here as well

KuznetsovIvan, 2022-02-21:
I did the same and have the same issue. But author of this method specified it in this comment: https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/65#issuecomment-857015177 

Maybe i was able to figure out what the problem is in missing libraries.

When i run 
```
cd /mnt/sdcard/hacks/rtsp-server/meta/service/
./framegrabber-mainstream &
```
I have the error:
```
./framegrabber: error while loading shared libraries: libshbf.so.0.1: cannot open shared object file: No such file or directory
```
In this issue: https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/18 said that for version MJSXJ05CM some people recompile framegrabber. Unfortunately, I don't know if this will help or not.

I found 2 repositories that may help somehow solve this problem with framegrabber:
https://github.com/telmomarques/mija-framegrabber/
https://github.com/telmomarques/x360h1080p-rtsp-server/

if someone could run RTSP on 4.0.9 firmware it was awesome.

dannylin0711, 2022-09-01:
Got solution.
You need first install the hack and setup ssh first.
`
cd /mnt/data/lib
cp libshbf.so libshbf.so.0.1
cp libshbfev.so libshbfev.so.0.1
`