telmomarques/xiaomi-360-1080p-hacks #87: mjsxj05cm-hacks FYI

vekonyat, 2021-07-14:
Anyone interested in this cam : mjsxj05cm
There are two working hacks I've found for the 3.5.1_0052 firmware version (IPC019_3.5.1_0052)
Nickbel, Russian guy, working RTSP, SSH and standard ONVIF PTZ, no web interface:
https://www.nickbel.com/2020/11/09/xiaomi-mi-smart-camera-360-1080p-sobstvennaya-proshivka.html#cut
Found this in brasilian Home Assistant forum, maybe exists here somewhere, but I couldn't find, RTSP, SSH, WEB, no PTZ?:
https://forum.homeassistantbrasil.com.br/t/cameras-compativeis-com-o-home-assistant-onvif-rtsp/144/89
It says, it can be used with 4.0.9_0425, but I presume it still should be downgraded to 3.5.1_0052.

If you have higher version, and can't downgrade, you should use the flasher and the clip to reflash the bios chip to 3.5.1_0052, that's what I did, not a big deal:
https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/18
I still couldn't downgrade to 3.4.2_0062, so I couldn't use this current hack from cmiguelcabral: https://github.com/cmiguelcabral/mjsxj05cm-hacks, but maybe I was just too lame for that.

Toukite, 2021-11-08:
Hello,
I managed to do this:

1. Reset the camera using this https://drive.google.com/open?id=1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj (link found on another issue after bricking my camera which was stuck yellow after trying to install this telmomarques hack)
2. Configure & install Nickbel hack

I confirm that ONVIF / RTSP / PTZ OVER ONVIF are working.
However, I have following issues:

- Camera does not detect night, so not usable at night unless I turn on the room light first
- Sound not available
- Unable to change some basic configurations (like reversing the image, change IR mode, ...)
- NTP is buggy, sometimes I have the correct value, sometimes it's completely off, sometimes the GMT+X config is not taken into account (a few hours different from expected)
- Unable to restore to step 1 --> camera does reset, but then it's impossible to re-add it to xiaomi, there is a timeout during connection. But camera is not fully bricked, just have to restart it again with Nickbel and we're back there (but we're now stuck with that solution).

If anyone can help on one of those issues, it would be appreciated :)

vekonyat, 2021-12-20:
> Hi there,
> 
> So you consider the Nickbel the best working solution?
> 
> Thanks for the help been struggling in finding information on this device and how to hack it

Hi, sorry for the late reply. Yes, that's the one I an using now with no issues.

closed 2021-12-20 by vekonyat