telmomarques/xiaomi-360-1080p-hacks #88: MJSXJ02CM Brick

pexcn, 2021-07-28:
Hello everyone:
Anybody can provide me MJSXJ02CM programmer firmware backup? My camera was brick, and i don't have a backup. 😭

albgau, 2021-07-28:
Hello,
I think, it could be done with the downgrade proc
https://github.com/telmomarques/xiaomi-360-1080p-hacks#downgrade-the-firmware

pexcn, 2021-07-28:
> Hello,
> I think, it could be done with the downgrade proc
> https://github.com/telmomarques/xiaomi-360-1080p-hacks#downgrade-the-firmware

Thank you for your reply.
In fact, I used CH341A to clear the flash, and it cannot be upgraded or downgraded, because the boot partition no longer exists.
Fortunately, I found a programmer backup here: https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/8#issuecomment-468950094
Now my device has been resurrected. 😄

closed 2021-07-28 by pexcn