telmomarques/xiaomi-360-1080p-hacks #91: Mi360 2k Pro MJSXJ06CM

tozzer, 2021-11-18:
In a similar vein has anyone had any luck with the MJSXJ06CM version of the camera? 
Mine came with 4.1.6 firmware installed, so I suspect downgrading to 3.x may not even be possible.

cvachta, 2021-12-22:
I have same camera. I have firmware version 4.1.6_0309. Camera **MJSXJ06CM** factory id **IPC021A04**. 

I have downloaded firmware from this http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/IPC021A04/firmware/IPC021A04_4.1.6_0309.zip

I'm not sure if the firmware can be unpacked, patched and reloaded without using a programmer - just with sdcard.

```
# Extract parts
dd if=tf_update.img of=files/header.bin bs=1 count=64 skip=0
dd if=tf_update.img of=files/xzdata.xz bs=1 count=2097088 skip=64
dd if=tf_update.img of=files/squashfs.bin bs=1 count=7733248 skip=2097152
dd if=tf_update.img of=files/jffs2.bin bs=1 skip=9830400

# Remove 0xFF from end of file
sed '$ s/\xff*$//' files/xzdata.xz > files/xzdata_without_pad.xz

# Unpack
xz -d files/xzdata_without_pad.xz
unsquashfs files/squashfs.bin
```

tomacoglu, 2022-01-17:
Hello, any progress?
Can we install it on Mi 360 2K, this hack?
thanks.

yokrysty, 2022-02-04:
my solution to not damage the board:
dump the NOR flash using a SOIC test clip
[image]
a little tool to read UART using POGO pins and a wooden clip
[image]
[image]

AAngold, 2022-02-06:
@yokrysty Can you list the equipment used for pulling the backup? (E.g. USB to SOIC Adaptor)

yokrysty, 2022-02-07:
@AAngold 
- SOP8 SOIC8 Test Clip and CH341A USB Programmer Flash for Most of 24 25 Series EEPROM BIOS Chip
- Ch341A_NeoProgrammer_2.2.0.10 using config for MX25L12875F or MX25L12873F because it's similar to MX25L12833F used in the camera

milhouse-dev, 2022-02-10:
@yokrysty have you been successful with downgrading? If so, I would be very thankful if you could provide me with instructions on how to accomplish it. Thanks in advance!

yokrysty, 2022-02-10:
I don't need downgrade, I am experimenting with OpenIPC and other personal development, but I don't see any reason why you can't downgrade when you connect directly to the NOR flash

milhouse-dev, 2022-02-10:
I understand. Sorry, but I'm new to this IP cam stuff. OpenIPC looks pretty interesting :) I was looking to downgrade in order to have web access to the Xiaomi mentioned above. I have ordered the hardware and will try to figure out how to flash. Shouldn't be rocket science I guess. Thanks for your reply!

Hi-hass, 2022-07-02:
I found an [older version](http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc021/IPC021_4.0.6_0132.zip)from [here](https://www.chinafix.com/thread-1305340-1-1.html), but its factory id is IPC021 not IPC021A04.
Can I downgrade to this firmware?

Hi-hass, 2022-07-02:
Now the latest version is 4.3.4_0372, I didn't find any link to download it. 
Does anyone find it?

delarqueen, 2022-08-21:
i need 4.3.4_0372 recovery file please help me

xTREMIST33, 2022-08-27:
> i need 4.3.4_0372 recovery file please help me

Has anyone find it? it keeps bricking when I try to update from XIAOMI HOME ME app

xTREMIST33, 2022-08-27:
Has anyone find  4.3.4_0372 recovery file / update file ? it keeps bricking when I try to update from XIAOMI HOME ME app

delarqueen, 2022-08-27:
> Has anyone find 4.3.4_0372 recovery file / update file ? it keeps bricking when I try to update from XIAOMI HOME ME app

I found and its work if you want to night i can attach

Antonisrrr, 2022-08-28:
i fix it with this file  http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/IPC021A04/firmware/IPC021A04_4.1.6_0309.zip

copy to sd car fat 32  tf_update.img leave the camera to rotate

sparkle9400, 2022-08-29:
can anyone attach the newest frimware for mjsxj06cm (2k Pro)?

yokrysty, 2022-08-30:
> can anyone attach the newest frimware for mjsxj06cm (2k Pro)?

For me its 4.1.6_0310. I am in EU.
I can make you a flash dump.
Let me know if that helps you.

thiagothomes, 2022-09-04:
> > can anyone attach the newest frimware for mjsxj06cm (2k Pro)?
> 
> For me its 4.1.6_0310. I am in EU. I can make you a flash dump. Let me know if that helps you.

Please, May someone know a reliable source to download latest firmware for mjsxj06cm (2k Pro)  ?
It looks like that my camera bricket itself updating to latest firmware (it is stucked with orange light  indicator) ...

dongFangTuring, 2022-09-07:
@yokrysty I need the flash dump.My camera is stucked with orange light indicator.thanks.

sparkle9400, 2022-09-07:
[Use this link to download the tf_update.img (to unbrick)](https://drive.google.com/file/d/10lV2FURnFubQ52K0-rfCOZ2dyehQbkun/view) @dongFangTuring

dongFangTuring, 2022-09-07:
> [Use this link to download the tf_update.img (to unbrick)](https://drive.google.com/file/d/10lV2FURnFubQ52K0-rfCOZ2dyehQbkun/view) @dongFangTuring
I put the file to sdcard,but not work.So i think use CH341A write other camera flash dump to flash chip maybe useful.

PecceG2, 2022-10-22:
Has anyone managed to install the hacks on the new model?
Or at least achieve a downgrade?
I am completely stuck in a camera that, at any moment, goes flying out the window.
Thanks!

tomacoglu, 2022-11-06:
I understand that cheats are not installed on the new model (MJSXJ06CM). I wish I had bought 1080p instead of 2K :)

mlaeng, 2022-12-31:
@yokrysty were you able to do anything with this cam?

yokrysty, 2023-01-01:
> @yokrysty were you able to do anything with this cam?

yes i explained a couple of posts above how
connecting directly to the flash

ckcr4lyf, 2023-02-20:
Just picked up the camera today, trying to see if I can make it work "offline" from Xiaomi. I am exploring it from a networking perspective. Since I run my own router in pure ubuntu/iptables etc, I ran `tcpdump` to capture the whole connection (including initial Wifi connection - see DHCP offer).

Unfortunately it seems that it uses TLS, so after the initial handshake there isn't much to see:

[image]

Interestingly though, it seems the SSL cert is self signed. I've attached the OpenSSL connection attempt:

<details>

<summary>
OpenSSL connection to pairing server
</summary>

```
$ openssl s_client -connect 8.219.193.159:443 -showcerts
CONNECTED(00000003)
Can't use SSL_get_servername
depth=0 O = Mijia Cloud, C = CN, CN = Mijia Cloud
verify error:num=20:unable to get local issuer certificate
verify return:1
depth=0 O = Mijia Cloud, C = CN, CN = Mijia Cloud
verify error:num=21:unable to verify the first certificate
verify return:1
depth=0 O = Mijia Cloud, C = CN, CN = Mijia Cloud
verify return:1
---
Certificate chain
 0 s:O = Mijia Cloud, C = CN, CN = Mijia Cloud
   i:O = Mijia Root, C = CN
   a:PKEY: id-ecPublicKey, 256 (bit); sigalg: ecdsa-with-SHA256
   v:NotBefore: May 23 06:28:50 2018 GMT; NotAfter: May 10 06:28:50 2068 GMT
-----BEGIN CERTIFICATE-----
MIIBjjCCATSgAwIBAgIBATAKBggqhkjOPQQDAjAiMRMwEQYDVQQKEwpNaWppYSBS
b290MQswCQYDVQQGEwJDTjAgFw0xODA1MjMwNjI4NTBaGA8yMDY4MDUxMDA2Mjg1
MFowOTEUMBIGA1UECgwLTWlqaWEgQ2xvdWQxCzAJBgNVBAYTAkNOMRQwEgYDVQQD
DAtNaWppYSBDbG91ZDBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABKdS7NRLazsX
q8NPgwDGMg8uTL7FelEDS17K33NH10XfjD28+q7bZ7BMrOWv95gYLkPFpES2J8LX
82FinT+RSAKjQjBAMB8GA1UdIwQYMBaAFJa3onw5sblmM6n40QmyAGDI5sURMB0G
A1UdDgQWBBRaKb/7L7dQDOnEIPI9iZtv4IAykzAKBggqhkjOPQQDAgNIADBFAiEA
khRZlgi2UbYZSoBSDiE5sUfdGwzVHXkxVFYlHygkdJACIAueLVcYN+Nqel5ginX3
l8cAHK+6lKHgOoQJ+XZ34a+t
-----END CERTIFICATE-----
---
Server certificate
subject=O = Mijia Cloud, C = CN, CN = Mijia Cloud
issuer=O = Mijia Root, C = CN
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: ECDH, secp521r1, 521 bits
---
SSL handshake has read 795 bytes and written 487 bytes
Verification error: unable to verify the first certificate
---
New, TLSv1.2, Cipher is ECDHE-ECDSA-AES128-GCM-SHA256
Server public key is 256 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-ECDSA-AES128-GCM-SHA256
    Session-ID: 0C9BB10A9C2176F43E7F6AA21B63C2A20BB2AC7BD90494FCDA7F2CC2464ACE45
    Session-ID-ctx:
    Master-Key: 5A099185AD4FED7F931A4E5A727D7164A9B672D322DCD7CE240561830D8E0D7834185DE1263267F32C6824F3A2DE542C
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    Start Time: 1676906073
    Timeout   : 7200 (sec)
    Verify return code: 21 (unable to verify the first certificate)
    Extended master secret: yes
---
```

</details>

I'm gonna play around with some SSL downgrade or similar stuffs.

ckcr4lyf, 2023-02-20:
Another thing I did is use iptables to prevent the smart camera from actually connecting to the internet (just drop all connections).

I can see several TCP SYN being sent to that pairing server, as some as some UDP messages to another one. Here are some UDP messages, I've no idea what it is trying to send:

```
15:22:02.637758 IP 192.168.30.108.54321 > 8.219.48.186.8053: UDP, length 32
	0x0000:  4500 003c 0000 4000 4011 2208 c0a8 1e6c  E..<..@.@."....l
	0x0010:  08db 30ba d431 1f75 0028 c979 2131 0020  ..0..1.u.(.y!1..
	0x0020:  ffff ffff ffff ffff 0000 07dd 6062 9bb0  ............`b..
	0x0030:  98ba 8ad0 dc2e 9eb1 0c59 59ce            .........YY.
15:22:04.978633 IP 192.168.30.108.54321 > 8.219.48.186.8053: UDP, length 32
	0x0000:  4500 003c 0000 4000 4011 2208 c0a8 1e6c  E..<..@.@."....l
	0x0010:  08db 30ba d431 1f75 0028 7ccc 2131 0020  ..0..1.u.(|.!1..
	0x0020:  ffff ffff ffff ffff 0000 07df 62b9 855c  ............b..\
	0x0030:  e476 527f be5c 218e ef0b 5f4e            .vR..\!..._N
15:22:07.464734 IP 192.168.30.108.54321 > 8.219.48.186.8053: UDP, length 32
	0x0000:  4500 003c 0000 4000 4011 2208 c0a8 1e6c  E..<..@.@."....l
	0x0010:  08db 30ba d431 1f75 0028 7e69 2131 0020  ..0..1.u.(~i!1..
	0x0020:  ffff ffff ffff ffff 0000 07e2 a469 015a  .............i.Z
	0x0030:  8e90 e1c9 3718 f187 e10a 2be8
```

Note: for UDP it tried to lookup the domain `sg.ot.io.mi.com` , as opposed for the TCP/TLS attempts it tried `sg.ots.io.mi.com`

ckcr4lyf, 2023-02-20:
I also tried blocking just TLS (via `iptables -A FORWARD -p tcp --dport 443 -j DROP` , and then attempt to pair. However it fails with network error, even though there is some UDP activity going on with `sg.ot.io.mi.com` . 

I've attached the pcap if anyone wants to try and take a look at the UDP stuffs (need to zip for Github)

[cam_block_tcp.zip](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/10785771/cam_block_tcp.zip)

ckcr4lyf, 2023-02-21:
So I got a bit further along - there was an issue with my mitmproxy because of which I couldn't perform an EC handshaek, but with that fixed, the camera rejects my MITM attempt:

[image]

I downloaded some of the firmware you guys shared, and ran `binwalk` on it. I discovered that apart from a standard linux filesystem, there is some addition stuff, specifically of interest to me is they include their "Fake CA"  public cert:

```
IPC021A04_4.1.6_0309/_tf_update.img-1.extracted/jffs2-root/data$ cat MijiaRootCert.pem
-----BEGIN CERTIFICATE-----
MIIBazCCAQ+gAwIBAgIEA/UKYDAMBggqhkjOPQQDAgUAMCIxEzARBgNVBAoTCk1p
amlhIFJvb3QxCzAJBgNVBAYTAkNOMCAXDTE2MTEyMzAxMzk0NVoYDzIwNjYxMTEx
MDEzOTQ1WjAiMRMwEQYDVQQKEwpNaWppYSBSb290MQswCQYDVQQGEwJDTjBZMBMG
ByqGSM49AgEGCCqGSM49AwEHA0IABL71iwLa4//4VBqgRI+6xE23xpovqPCxtv96
2VHbZij61/Ag6jmi7oZ/3Xg/3C+whglcwoUEE6KALGJ9vccV9PmjLzAtMAwGA1Ud
EwQFMAMBAf8wHQYDVR0OBBYEFJa3onw5sblmM6n40QmyAGDI5sURMAwGCCqGSM49
BAMCBQADSAAwRQIgchciK9h6tZmfrP8Ka6KziQ4Lv3hKfrHtAZXMHPda4IYCIQCG
az93ggFcbrG9u2wixjx1HKW4DUA5NXZG0wWQTpJTbQ==
-----END CERTIFICATE-----
```
Since it is actually checking the signature, my nooby attempts to MiTM won't work directly...

> copy to sd car fat 32 tf_update.img leave the camera to rotate
> Thanks

@Antonisrrr were you able to use this to "reload" firmware onto the device? I am wondering if I could patch the current firmware to add my custom CA, then I can sniff all the camera's traffic, which would greatly help in my attempts to reverse engineer their network protocol side of things

PecceG2, 2023-02-21:
Thank you very much in advance for all the work you are contributing @ckcr4lyf .
I'm not very good with low-level programming, but I have a camera on my desk and electronics/programming tools and I'm willing to help in any way I need.

ckcr4lyf, 2023-02-22:
Seems like passive recon is a no go... I tried with a valid cert but incorrect Common Name (just had lets encrypt issue a cert for a domain I own), and unfortunately the camera won't accept it.

[image]

I will attempt to link the camera normally and just try to analyze the non UDP traffic instead.

Stif007, 2023-04-20:
I found some good stuff on the internet! maybe he will help you? https://home.miot-spec.com/spec/chuangmi.camera.021a04

ckcr4lyf, 2023-04-20:
> I found some good stuff on the internet! maybe he will help you? https://home.miot-spec.com/spec/chuangmi.camera.021a04

What exactly is this @Stif007 ? Unable to figure out if its some kind of API docs / request format

djangobits, 2023-04-21:
@ckcr4lyf seems to be for this https://github.com/al-one/hass-xiaomi-miot

P3run, 2023-05-18:
@yokrysty I would really appreciate I you could share your dump with me. I haven't been able to flash my camera with SD card so my only hope is to push the dump directly to memory.

yokrysty, 2023-05-19:
@P3run hello, i cannot give you the full dump, because there is some private data in some sections (config, factory)
i will give you the rest of them and you need to reconstruct the image with your private sections

the flash image layout is:
```
0x000000 0x0050000 boot
0x050000 0x0250000 kernel
0x250000 0x09b0000 rootfs
0x9b0000 0x0fe0000 data
0xfe0000 0x0ff0000 config
0xff0000 0x1000000 factory

```
[backup.zip](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/11515272/backup.zip)

ckcr4lyf, 2023-05-19:
By connecting directly to the chip, is it possible to overwrite the flash?

I am thinking if it might be worth investing in some wires / pins etc. tooling to do that, then I can add my own CA to the firmware to analyze the TLS traffic (since they hardcode the root cert in their firmware -> https://github.com/telmomarques/xiaomi-360-1080p-hacks/issues/91#issuecomment-1438620756)

Supremo1119, 2023-05-26:
-- Can someone help me, i think i brick my cam while updating i accidentally unplug it's power.
Model: mi 360 home security camera 2k pro (mjsxj06cm)
Serial: 28309
Problem: stuck yellow light

Thank you from Philippines  :)

P3run, 2023-05-26:
> @P3run hello, i cannot give you the full dump, because there is some private data in some sections (config, factory) i will give you the rest of them and you need to reconstruct the image with your [...]

Thank you so much! I'll try to fix it this weekend :).

hyunho-yoon, 2023-06-01:
So far, no one has succeeded in Hack or RTSP in Mi360 2K PRO MJSXJ06CM, right?

IulianMm, 2023-06-11:
> -- Can someone help me, i think i brick my cam while updating i accidentally unplug it's power. Model: mi 360 home security camera 2k pro (mjsxj06cm) Serial: 28309 Problem: stuck yellow light
> 
> Thank you from Philippines :)

Hello @Supremo1119 !
Did you managed to solve the problem with the camera?
If yes, what were the steps you have taken?

ckcr4lyf, 2023-06-12:
> unzip this and copy to tf card

@firedevel do you know if the firmware is signed? If I modify it a bit, would it sitll work if I try and update it?

IulianMm, 2023-06-14:
> > > > -- Can someone help me, i think i brick my cam while updating i accidentally unplug it's power. Model: mi 360 home security camera 2k pro (mjsxj06cm) Serial: 28309 Problem: stuck yellow light
> > > > Thank you from Philippines :)
> > > 
> > > 
> > > Hello @Supremo1119 ! Did you managed to solve the problem with the camera? If yes, what were the steps you have taken?
> > 
> > 
> > unzip this and copy to tf card http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/IPC021A04/firmware/IPC021A04_4.1.6_0309.zip
> 
> Hello @firedevel ! Thank you for the information you've shared. Sadly, didn't worked on my device. I still have my camera blocked with the yellow/orange light. I see version for the folder is "4.1.6_0309". Does this come from the firmware version? Looking in my app i see the my firmware is 4.3.4_0372. Also, I'm using a 64GB SD card for the camera.
> 
> Thanks again for your help!

Problem solved! 
The reason why the provided image didn't "do the trick" was because I was using a god damn 64 card for the camera. I switched to a 16GB card where I had the option to format it in FAT32. 
I used the image from the IPC021A04_4.1.6_0309 folder and the magic happened! 

@firedevel , thanks a lot for your help!

HC23393, 2023-08-18:
Thanks to Firedevel...my Mi home security 360 2K Pro..,MJSXJ106
S)N 28309.   Stuck at orange light.
Flashed Bios IPC021A04_4.1.6_0309.
Firstme worked... everything intact.18.Aug.2023.
I m from Malaysia.

javi0701, 2023-11-23:
Hola mi camara mi 360° de xiaomi, no conecta y la luz es naranja fija y aveces intermitente. Me podrian ayudar intente con el finware IPC021A04_4.1.6_0309. y no funciono. Sospecho que necesito una version mas antiigua.
Quedo atento.

firedevel, 2023-11-24:
> Thanks to Firedevel...my Mi home security 360 2K Pro..,MJSXJ106 S)N 28309. Stuck at orange light. Flashed Bios IPC021A04_4.1.6_0309. Firstme worked... everything intact.18.Aug.2023. I m from Malaysia.

Sorry, the firmware is for MJSXJ06CM A04 (SN: 30486*) and not for SN 28309.  
there is another firmware for MJSXJ06CM (SN: 26537*): [https://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc021/IPC021_4.0.9_0178.zip](https://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/ipc021/IPC021_4.0.9_0178.zip) hope it helps.

See my repository for more details: [Xiaomi-Camera-firmware](https://github.com/firedevel/Xiaomi-Camera-firmware)

firedevel, 2023-11-24:
I found an older firmware here: [http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/IPC021A04/firmware/IPC021A04_4.1.2_0282.rar](http://cdn.cnbj2.fds.api.mi-img.com/chuangmi-cdn/product/IPC021A04/firmware/IPC021A04_4.1.2_0271.rar)

https://github.com/firedevel/Xiaomi-Camera-firmware/tree/main/MJSXJ09CM-A04

psnlk, 2024-03-14:
MJSXJ06CM anyone connect this to frigate?

psnlk, 2024-03-14:
> So far, no one has succeeded in Hack or RTSP in Mi360 2K PRO MJSXJ06CM, right?

Are you?

srj24x7, 2024-04-19:
Does anyone have firmware for MJSXJ06CM SN: 31957 ? It was purchased in India and has stopped working, stuck at yellow light. I could successfully flash it with IPC021A04_4.1.2_0282.rar but still I am unable to connect the camera to Mi Home or the Wi-Fi. Please help.

931122, 2024-07-31:
> So far, no one has succeeded in Hack or RTSP in Mi360 2K PRO MJSXJ06CM, right?

i have got rtsp

djangobits, 2024-07-31:
@931122 Wat? How?

Nickolay40, 2024-08-01:
> > So far, no one has succeeded in Hack or RTSP in Mi360 2K PRO MJSXJ06CM, right?
> 
> i have got rtsp

instructions, at least short?

931122, 2024-08-06:
> @931122 Wat? How?

https://github.com/931122/xiaomi_mjsxj06cm_rtsp_server.git

ckcr4lyf, 2024-08-06:
Pretty interesting, thanks for sharing.

For others wondering, this does need hardware access to upload the flash: https://github.com/SungurLabs/sungurlabs.github.io/blob/6043366d497943e0a246a6a420ba8fb2adfcef31/_posts/2021-07-14-Xiaomi-Smart-Camera---Recovering-Firmware-and-Backdooring.md#firmware-backdooring

Still, at least with some hardware work it is possible!

Nickolay40, 2024-08-06:
> > @931122 Wat? How?
> 
> https://github.com/931122/xiaomi_mjsxj06cm_rtsp_server.git
you wrote.. you can find uart pins - on photo below:
[image]

931122, 2024-08-06:
> > > So far, no one has succeeded in Hack or RTSP in Mi360 2K PRO MJSXJ06CM, right?
> > 
> > 
> > i have got rtsp
> 
> instructions, at least short?

I didn't write a detailed tutorial, but once you get the root, you can run rtsp_demo.

Nickolay40, 2024-08-06:
> > > > So far, no one has succeeded in Hack or RTSP in Mi360 2K PRO MJSXJ06CM, right?
> > > 
> > > 
> > > i have got rtsp
> > 
> > 
> > instructions, at least short?
> 
> I didn't write a detailed tutorial, but once you get the root, you can run rtsp_demo.

Yes, rtsp server working, but only after manual run. 
May be, I wrote some think wrong?
[image]

931122, 2024-08-07:
> > > > > So far, no one has succeeded in Hack or RTSP in Mi360 2K PRO MJSXJ06CM, right?
> > > > 
> > > > 
> > > > i have got rtsp
> > > 
> > > 
> > > instructions, at least short?
> > 
> > 
> > I didn't write a detailed tutorial, but once you get the root, you can run rtsp_demo.
> 
> Yes, rtsp server working, but only after manual run. May be, I wrote some think wrong? ![Screenshot from 2024-08-06 14-36-47](https://private-user-images.githubusercontent.com/31587809/355450289-66c7eb96-3ff6-4ebd-8da5-cb0165cdab01.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI5OTM1NjIsIm5iZiI6MTcyMjk5MzI2MiwicGF0aCI6Ii8zMTU4NzgwOS8zNTU0NTAyODktNjZjN2ViOTYtM2ZmNi00ZWJkLThkYTUtY2IwMTY1Y2RhYjAxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDAxMTQyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU3M2Q3NjRiY2U3ZDIyNDdjNGU5MTlkNzlhZDY4ZGIxMDhiNmI5ZmJjOWJkZGJmYTNmZWI2MjUwNzkyZGI1OTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IvMteEoVOv8sXf1F4T8BFu0OAJlcgNPpn_DjmqK7V1g)

You should run it after mi_stream， so
sleep 90 && mi_rtsp_server &

hjelmdal, 2024-08-23:
@931122 , can this be installed / flashed from an SD card without any hardware, or do we need to physically tamper with the camera?

[3 more replies]