telmomarques/xiaomi-360-1080p-hacks #94: MJSXJ10CM support

fantnhu, 2022-01-07:
Hi!
Please make support for MJSXJ10CM model

8999400, 2022-01-09:
BHR4885GL/MJSXJ10CM

mhackdo18, 2022-05-08:
+1 here is it compatible?

FrancoisT44, 2022-12-30:
Hi
I'm reviving this somewhat topic it might be of interest to me but i want just to be sure .  I have a Xiaomi Mi 360 1080p (MJSXJ10CM).
I nevertheless intend to try with (MJSXJ02CM) do you think that I risk for my camera?
If so, what are the possible ways to use this camera without xiaomi interface 
thank you for your comeback
fr

NicolasGuilloux, 2023-01-07:
Hello!

I think at this point, it is fair to say that no solution will come from this repository. It has no commit activity for 2 years and the maintainer didn't even reply to this issue.
The sad part is that apparently nobody knows how to port it to the MJSXJ10CM, or at least nobody that knows showed an interest whatsoever.

I'm planning to exchange my camera with a friend that has an MJSXJ02CM. Having my camera feed going on a server somewhere in China is unacceptable, especially because it is an indoor one. Moreover, it doesn't not work with HomeAssistant, so it remains unplug for now :(

Hope somebody will find a solution to this hassle