telmomarques/xiaomi-360-1080p-hacks #95: MJSXJ05CM downgrade from 3.5.1_0052

jcadel, 2022-01-11:
Hi guys,

I tried to downgrade my MJSXJ05CM camera from 3.5.1_0052 to 3.4.2_0062 firmware but it didn't work.
The tf_recovery.img that I used was the one which can be found at https://github.com/cmiguelcabral/mjsxj05cm-hacks
My microSD card is a Sandisk Class 10 64Go formatted in FAT32.

Did any of you succeed ? If so, would you please tell me which microSD and recovery file you were working on ?

Thanks in advance !

freezir12, 2022-01-12:
Would like to know too! Waited for hours but light didnt turn blue, it stayed orange.

GaetanPTX, 2022-01-15:
Hi! I have the same problem, it's as if my SD card was ignored by the camera.

freezir12, 2022-01-16:
if you downgrade is not working, then use the hack-files from here (do not need downgrade from 3.5.1_0052) : https://github.com/zry98/MJSXJ05CM-hacks
an alternative would be using the hack from nickbel (do not need downgrade from 3.5.1_0052) : https://www.nickbel.com/2020/11/09/xiaomi-mi-smart-camera-360-1080p-sobstvennaya-proshivka.html#cut

what i did everytime before i have put in the SD Card with hack files on it: 
bringing the device back to factory standards by using the "unbrick" tf_recovery.img (https://drive.google.com/file/d/1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj/view)  and disconnect it from the xiaomi app.
Basically i know my device is ready to be hacked and ready to read the files from the SD Card when the device is in the state of and shouting :" Waiting for connection".

maxxe88, 2022-01-16:
the nickbel version worked for me. but i dont get any stream. when i connect to the camera with the ONVIF Device Manager I can move the camera but I dont get any video. can anyone help?

GaetanPTX, 2022-01-16:
> if you downgrade is not working, then use the hack-files from here (do not need downgrade from 3.5.1_0052) : https://github.com/zry98/MJSXJ05CM-hacks an alternative would be using the hack from nickbel (do not need downgrade from 3.5.1_0052) : https://www.nickbel.com/2020/11/09/xiaomi-mi-smart-camera-360-1080p-sobstvennaya-proshivka.html#cut
> 
> what i did everytime before i have put in the SD Card with hack files on it: bringing the device back to factory standards by using the "unbrick" tf_recovery.img (https://drive.google.com/file/d/1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj/view) and disconnect it from the xiaomi app. Basically i know my device is ready to be hacked when the device is in the state of and shouting :" Waiting for connection".

Thank you for all these tips! I looked at the nickbel hack page. I do all the procedure indicated but when I put on the SD card and then I start the camera. The camera does not boot the hack, I waited 30 minutes for the led to turn blue but the led still flashes orange.

freezir12, 2022-01-17:
> the nickbel version worked for me. but i dont get any stream. when i connect to the camera with the ONVIF Device Manager I can move the camera but I dont get any video. can anyone help?

was the same for me. i do not think you will get a videostream over ONVIF, i only used it to control the cameras motor. i got a videostream via rtsp://<webcam-IP>:8554/mainstream    only. (maybe "substream" instead of "mainstream" works too.

freezir12, 2022-01-17:
> > if you downgrade is not working, then use the hack-files from here (do not need downgrade from 3.5.1_0052) : https://github.com/zry98/MJSXJ05CM-hacks an alternative would be using the hack from nickbel (do not need downgrade from 3.5.1_0052) : https://www.nickbel.com/2020/11/09/xiaomi-mi-smart-camera-360-1080p-sobstvennaya-proshivka.html#cut
> > what i did everytime before i have put in the SD Card with hack files on it: bringing the device back to factory standards by using the "unbrick" tf_recovery.img (https://drive.google.com/file/d/1ve6XlBEiZebJV6ukJ0Oiu7DePw2JCsWj/view) and disconnect it from the xiaomi app. Basically i know my device is ready to be hacked when the device is in the state of and shouting :" Waiting for connection".
> 
> Thank you for all these tips! I looked at the nickbel hack page. I do all the procedure indicated but when I put on the SD card and then I start the camera. The camera does not boot the hack, I waited 30 minutes for the led to turn blue but the led still flashes orange.
> 
> The camera remains stuck on "waiting for connection"

sorry have no time right now to do the nickbel procedure again. i could do it tomorrow and provide more information. but for now some thoughts (maybe you already did/follow these points, just asking to be sure): 
- is your  SD Card FAT formated?
- try to use SD Card with 16 GB max
- did you disconnect your camera from the app?
- did you "reset" your camera with the tf_recovery.img?
- is your default manifacture firmware 3.5.1_0052?
- did you put the "manu_test" folder on the cards root directory?

GaetanPTX, 2022-01-17:
Hello, thank you for your answer, here are the answers to your questions below 

- The card is formatted in FAT 32
- I tried with a 64GB card and another 8GB card
- yes, the camera is disconnected from the application
- I can't downgrade the camera version, my version is 4.0.9_0426 (every time I start the downgrade procedure, my camera ignores the SD card and starts normally)
- How do I know the version of my default firmware?
- Yes, I put the contents of the manu_test folder in the SD card
- Yes, my camera model is: MJSXJ05CM

Thank you for your help

freezir12, 2022-01-17:
The difference between our setups is the firmware. Yours is 4.0.9_0426, mine is 3.5.1_0052. I didnt have to downgrade, you have to downgrade, but as far as i know it is only possible by dismantling the camera and use an spi flasher. I am still waiting for my flasher module i have ordered on amazon (cause i wanna go even lower than 3.5.1_0052) so i can not provide a lot information on this. please check out other github projects about those cameras and their issue section.

VictorPiella, 2022-10-10:
Hi guys, I have tried everything you say and there is no way it works. I haven't tried the nickbel hack as I wanted to keep using my home. My current status with or without sd is the same. It turns on with the yellow led and after a while it turns blue and moves the motors. It doesn't play any audio, I tried changing the speaker, but it doesn't respond with a new one either. When trying to configure with the app it never scans the code on the screen (I also tried to print it and nothing). I have checked that the usb connector is a bit burnt, but when I connect the motors and the led they work. Any ideas ?