telmomarques/xiaomi-360-1080p-hacks #104: MJSXJ14CM camera firmware dump

Regressor, 2023-09-14:
I got a Xiaomi C200 camera (MJSXJ14CM) opened it and downloaded a dump from 25vq128 spi nand. It looks like they hardened factory mode. Now it looks like this:

```
        if [ -f "${MANUFA_DAT}" ];
        then
                rm -rf /tmp/manu_test
                rm -rf ${TMP_BIN}
                rm -rf ${TMP_DAT}
                cp ${MANUFA_BIN} /tmp/
                cp ${MANUFA_DAT} /tmp/
                cd /tmp/
                if [ $RSA_TYPE -eq 1 ];then
                        imi_echo "==============rsa:openssl================="
                        rm -fr md5sum1.txt
                        openssl rsautl -verify -in "${TMP_DAT}" -inkey /mnt/data/data/cfg/public-key.pem -pubin -out md5sum1.txt
                        fact_md5=`md5sum -c md5sum1.txt`
                        if [ "${fact_md5}" = "manu.bin: OK" ] ; then
                                imi_echo "md5 ok"
                                rm -fr ${TMP_BIN}
                                tar -xf "${MANUFA_BIN}" -C /tmp/
                                cd /tmp/manu_test/
                                chmod -R 755 *
                                if [ -f "/tmp/manu_test/manu.sh" ]; then
                                        /tmp/manu_test/manu.sh
                                        if [ $? -ne 0 ]; then
                                                return 1
                                        else
                                                return 0
                                        fi
                                else
                                        return 1
                                fi
                        else
                                imi_echo "md5 fail"
                        fi

```

So it checks rsa sig before running manu.sh :(
I'll try to replace key inside squashfs and check if this helps...
Attached dump .bin file (renamed to log)

[Xiaomi_c200_dump.log](https://github.com/telmomarques/xiaomi-360-1080p-hacks/files/12607396/Xiaomi_c200_dump.log)

Regressor, 2023-11-15:
I desoldered flash, downloaded firmware, changed script, uploaded changed firmware and now it allow to run any script

danielopereira, 2023-12-06:
Hello @Regressor, 
could you share the files you used to flash it? 
Also if you could give us an help on how to do it, it would be really appreciated, i have same camera, and i would like to integrate it on HA.
Thanks

Regressor, 2023-12-08:
> Hello @Regressor, could you share the files you used to flash it? Also if you could give us an help on how to do it, it would be really appreciated, i have same camera, and i would like to integrate it on HA. Thanks

Currently I stuck on compiling rtsp/drive applications. There was a lack of free time because of work so it stays without any movement for a month or so.

tovy14, 2024-03-02:
Hi @Regressor did you get the time to make a how to? I think everyone would appreciate it

Regressor, 2024-08-05:
Ok. I got a freeeeeee time and spent a week trying to get it working. I made a docker image for cross compilation and adapted a  https://github.com/thewh1teagle/xiaomi-1080-360-motor-control.git code and it works - I can control motor thru small web page. Then I tried to build streamer (using ingenic sdk). Spent 2 days and then found OpenIPC web site.

I copied squashfs file to sdcard. Mounted it. Mounted (--bind) /dev /proc /tmp /run and some other dirs (including copied /etc to make it writable). Then I made chroot there, exported SENSOR=gc2083 and run majestic. There is no motor, but I can setup gpio pins for ircut, nitght light, night mode, speaker. It just works: I tested rtsp, hls, onvif, snapshots, audio (in and out), video streams, mjpeg and motion detection. OSD not working - core dump.

Still can't figure out how to setup night mode sensor and trying to add motor to it. Then I plan to make full openipc image to flash.

931122, 2024-08-06:
> Ok. I got a freeeeeee time and spent a week trying to get it working. I made a docker image for cross compilation and adapted a https://github.com/thewh1teagle/xiaomi-1080-360-motor-control.git code [...]

hi, I want to know how to make audio work

Regressor, 2024-08-06:
> hi, I want to know how to make audio work

OpenIPC majestic is not open-source :( But it work. Video, microphone and speaker. To get speaker you should setup speaker enable gpio pin 63.

I found this gpio map on MJSXJ14CM:

gpio8 - yellow led
gpio9 - blue led
gpio49 - night mode light leds
gpio50 - disable wifi ?
gpio 51 & gpio 54 - ircut drive
gpio63 - enable/disable speaker

Sensor gc2083 (i2c 0x37)
Processor Ingenic T31L

PTZ is controlled using ioctl calls on /dev/motor

Regressor, 2024-08-30:
Final news. I was able to run Thingino on C200 using some hacks. Xiaomi used secureboot to prevent firmware change so it wont boot alien uboot / kernel.

The only way is to solder arduino to uart to automate sf read xxx\;go xxx to execute new uboot over old one and then it boot thingino system. The uart points to solder on C200 board has a size of 0.2mm so if your solder skill not good - throw away this piece of shit (Xiaomi C200) and buy new one from another manufacturer.